dss-extensions/dss_capi #8: Can't find unit ESPVLControl used by DSSClassDefs

kdheepak, 2018-07-30:
I'm trying to compile on macos using the script in the package and it fails at the following step:

```
...
Compiling electricdss-src/Source/Controls/UPFCControl.pas
Assembling (pipe) build/units_v7_x64/UPFCControl.s
DSSClassDefs.pas(134,6) Fatal: Can't find unit ESPVLControl used by DSSClassDefs
Fatal: Compilation aborted
Error: /usr/local/bin/ppcx64 returned an error exitcode
/Library/Developer/CommandLineTools/usr/bin/objdump: 'lib/darwin_x64/v7/libdss_capi_v7.dylib': No such file or directory
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: can't open file: lib/darwin_x64/v7/libdss_capi_v7.dylib (No such file or directory)
error: /Library/Developer/CommandLineTools/usr/bin/install_name_tool: can't open file: lib/darwin_x64/v7/libdss_capi_v7.dylib (No such file or directory)
```

PMeira, 2018-08-16:
@kdheepak I think the current version is building fine without the changes. I created a branch to [build with FPC 3.0.0 on Travis-CI](https://travis-ci.org/PMeira/dss_capi/builds/416923621) to check if it was something with older FPC and it still worked:
```
Compiling /Users/travis/build/PMeira/electricdss-src/Source/Controls/UPFCControl.pas
Assembling (pipe) build/units_v7_x64/UPFCControl.s
(...)
Linking lib/darwin_x64/v7/libdss_capi_v7.dylib
(...)
Compiling /Users/travis/build/PMeira/electricdss-src/Parallel_Version/Source/Controls/UPFCControl.pas
Assembling (pipe) build/units_v8_x64/UPFCControl.s
(...)
Linking lib/darwin_x64/v8/libdss_capi_v8.dylib
```

I had to check how to install the official FPC DMG since v3.0.0 was not available on brew. Looks like it's faster than using the brew version (brew needs to update itself every time), so I'll remove brew from the travis.yml of the master branch:
```
          wget https://sourceforge.net/projects/freepascal/files/Mac%20OS%20X/3.0.0/fpc-3.0.0.intel-macosx.dmg/download -Ofpc.dmg
          sudo hdiutil attach fpc.dmg
          sudo installer -package /Volumes/fpc-3.0.0.intel-macosx/fpc-3.0.0.intel-macosx.pkg -target /
```

kdheepak, 2018-08-17:
Okay great! I'll close my PR. I'll also try this on my local machine without my conda install of freepascal.

closed 2018-08-17 by kdheepak