dss-extensions/dss_capi #10: MATLAB bindings

PMeira, 2018-08-13:
MATLAB offers two alternatives for interfacing external C-API libs:

- MEX
- CallLib/LoadLibrary

Using MEX is like using the CPython internal API, but worse in several aspects. CallLib/LoadLibrary is analogous to Python's CFFI -- just like in DSS Python, we can use a tweaked C header in LoadLibrary and it will build a specialized DLL automatically (provided there is a compiler). 

That specialized DLL can be redistributed, so end users should not need a C compiler. Just like with CFFI, the main difficulty will be building this. I still have no idea if a DLL built with older MATLABs would work on newer versions and vice-versa. If it doesn't work, distributing binaries the last two or three versions should be enough.

Using LoadLibrary and CallLib as the backend, the rest of the code should be pretty similar to DSS_Python and OpenDSSDirect.py. As soon as I have some free time, it should be a quick task now.

A new repository will be created to host the MATLAB bindings -- DSS_MATLAB? Maybe something else since it could include both a COM-like interface and an interface like OpenDSSDirect.py/.jl. 

@kdheepak When you have some time, if you have any specific requirements about this, feel free to let me know.

kdheepak, 2018-08-14:
Great! Thanks for deciding to tackle this! 

I like the idea of building `dss_matlab` to tackle the bindings first and then including a COM-like/odd interface later. 

I would prefer we used calllib/loadlibrary, because compiled MATLAB does not work with MEX. I also think if people want to generate the MEX file, they can do so (fairly) easily using swig, given that you already have header files that expose the C-ABI.

One challenge is how to create a test suite like that can be run on something like Travis. Do you know if Octave will work with calllib/loadlibrary? If so, we might be able to run Octave on Travis for testing purposes.

PMeira, 2018-08-14:
> I would prefer we used calllib/loadlibrary, because compiled MATLAB does not work with MEX.

That's good to know.

> One challenge is how to create a test suite that can be run on something like Travis. Do you know if Octave will work with calllib/loadlibrary? If so, we might be able to run Octave on Travis for testing purposes.

No, unfortunately Octave doesn't support those yet, they are still listed here: http://hg.savannah.gnu.org/hgweb/octave/file/88e87c69c8b8/scripts/help/__unimplemented__.m

PMeira, 2018-11-29:
@kdheepak Just made the repo public:

https://github.com/PMeira/dss_matlab

This is based on `calllib`. It seems the thunk DLL can be used across different MATLAB versions without issue. Most of the issues I found while porting to MATLAB were limitations on how pointers are handled.

I plan to test on Linux and Mac soon. Currently it reproduces the COM-like interface (I'd say 90% of the code is generated from the DSS Python code), we can add ODDpy-like interface later if there's interest. Most of the API should be covered already, some special cases (like `DSSProperty`) will need some changes.

> One challenge is how to create a test suite that can be run on something like Travis. 

I opened an issue about that -- https://github.com/PMeira/dss_matlab/issues/1

Independent of the CI system, I imagine custom workers (if allowed) with MATLAB installed on each platform would be enough.

closed 2018-11-29 by PMeira