dss-extensions/dss_capi #16: Add binaries back to github releases

kdheepak, 2019-01-30:
Is it possible to add the shared libraries back to github on tagged releases? This way it'll be easy for someone to download just the shared libraries they need. This will be particularly useful for OpenDSSDirect.jl at the moment.

kdheepak, 2019-01-30:
There are also many command line applications that let you upload a zip file to a github release, and you can set it up such that travis or appveyor compiles and pushes to a particular tag. For example, I have [a cli](https://github.com/kdheepak/github-release) too which pushes it's own releases to github through travis.

PMeira, 2019-01-30:
> Is it possible to add the shared libraries back to github on tagged releases? 

Sure, it shouldn't be too much trouble, I'll try to do something about it by tomorrow. I stopped providing binaries because most requests were for Windows binaries, which are always uploaded as AppVeyor artifacts anyway, but now binaries for all platforms will certainly be useful to automate other projects -- besides OpenDSSDirect.jl, we can use them for the C# and MATLAB too. Since you already managed to address Julia (thank you!), I'm considering adding some Java support soon, especially since COM usage with Java is more broken than with MATLAB.

The build process of DSS Python is already automated on tags, uploading to Anaconda cloud (wheels and conda packages), although I prefer to upload wheels to PyPI manually. It rebuilds several versions of the KLUSolve library with the MSVC compilers on Windows according to each Python version, and also because the manylinux wheels use different compilers from the conda-build environment. Even though I got repeated requests for conda packages, there is barely any download from the conda channel, so I might drop them in the future and simplify the whole process.

tshort, 2019-01-31:
Julia often runs into problems when libraries are built with MSVC compilers. On Windows, Julia is not built with MSVC. That can cause issues with libraries compiled with MSVC. I'm not sure if there's an issue here.

kdheepak, 2019-01-31:
If we are using the C standard interface (`cdecl`), we should be fine, no?

kdheepak, 2019-01-31:
I tested the Julia version on Windows using the Anaconda libraries and all the tests pass.

tshort, 2019-01-31:
That's probably right, @kdheepak. Maybe you only get into trouble with C++.

PMeira, 2019-01-31:
For DSS C-API itself, MSVC is (basically) only used for building KLUSolve, which exposes a C interface. The only detail is that we would assume the users have the MSVC runtime installed, which should be fine for Windows 10 at least, not sure for Windows 7. We probably need some tweaks in the `CMakeLists.txt` for using GCC or Clang on Windows, but otherwise should work fine -- I can do that later if we run into any issues, it's a good idea anyway.

The issue with Python is that each official version is build with an specific version of MSVC and we should match that for C extension modules (either the CFFI or the Cython version), so I also rebuild KLUSolve for each version for consistency. There are some tricky ways to build extensions with mingw compilers but usually not worth the effort.

By the way, I noticed the Julia distribution seems to include at least part of SuiteSparse, so we can probably build a leaner version later of KLUSolve later.

PMeira, 2019-02-01:
Done in #18, waiting a bit to merge just in case I'm missing something. As soon as I track (and test) recent changes in the official SVN repo, I'll tag 0.10.1.

@kdheepak I made a [pre-release with test binaries](https://github.com/PMeira/dss_capi/releases/tag/0.10.1a5), please let me know if you need any changes.

kdheepak, 2019-02-01:
Thanks for making the pre-release! Works like a charm for Mac, Linux and Windows!

PR #18 (2019-02-03): Automatic binary deployment on tags
Closes #16.
closed 2019-02-03 by PMeira