dss-extensions/dss_capi #23: Segfaults if DSSProperty_Set_Val(Value) is called without calling DSSProperty_Set_Index(argIndex)

kdheepak, 2019-02-05:
In the long run, it would be great to never have any function segfault. Additionally, it would be nice to have a better error reporting mechanism in the C API. Either: 

1) Have every function return an error code
2) Have every function take in a pointer to an struct that describes the error code / error description.
3) Populate a global error struct that is accessible through a separate function exposed in the CAPI.

This is low priority / nice to have. If we did this, in specific language bindings we can raise appropriate error signals. For example, in Python we can `raise DSSCAPIException(error_msg)` with a better error message to make the interface more Pythonic.

PMeira, 2019-02-06:
For the current incarnation, which tries to not deviate too much from the COM API conventions and keep the code in line with upstream, I think we have to go with:

>    3\. Populate a global error struct that is accessible through a separate function exposed in the CAPI.

The official OpenDSS already does this mostly through `Error_Get_Number` and `Error_Get_Description`, but it varies a lot. Besides, many functions don't have any error checking and/or error signaling as you noticed.

Along with `Error_Get_*`, some functions use a direct error/stop convention, such as returning 0 or -1. Another pair of functions uses an Pascal out parameter for the error code, `CktElement_Get_Variable`/`CktElement_Get_Variablei`, which I only fixed in December: https://github.com/PMeira/dss_capi/commit/12092385206da85362e1161d10dc8dff00664cfc

The new `Error_Get_NumberPtr` (not present in 0.10.0, will be out in 0.10.1) should help reduce the number of calls to `Error_Get_Number`, since we can reuse the pointer to the global variable and just check its contents. I'm slowly adding more errors that use this convention, so I'll check if `DSSProperty_Set_Val` is easy to handle.

If we were to change that in the future, I'd probably go with 1 -- either use it in the internals too or just wrap with try/except and map to error codes at the API level. The issue is that the current convention is used throughout the OpenDSS code and it would be easier to address that in a potential rewrite in the future.

PMeira, 2019-02-07:
Another issue with OpenDSS as a whole is that a lot of errors that can lead to misleading results are silently ignored. The work I'm doing on avoiding the DSS language parser for lots of things should help in the future (in a couple of months).

`DSSProperty_Set_Val`  just writes a command string that gets parsed later:

```pascal
//...
DSSExecutive.Command := 'Edit ' + ParentClass.Name + '.' + Name + ' ' + ParentClass.PropertyName^[FPropIndex] + '=' + String(Value);
```

(With this current implementation, might as well write that directly to the Text interface)

The issue at hand was that `FPropIndex` was not properly checked.

commit 52bc52b1 (2019-02-07): Validate the current property index in DSSProperty_Set_Val before using it, and raises error message on invalid values

Closes #23
closed 2019-02-07 by PMeira