dss-extensions/dss_capi #24: Make casing consistent for SInterval

kdheepak, 2019-02-06:
Currently this is in the header file:

```
    DSS_CAPI_V7_DLL double LoadShapes_Get_sInterval(void);
    DSS_CAPI_V7_DLL void LoadShapes_Set_Sinterval(double Value);
```

It can be changed to:

```
    DSS_CAPI_V7_DLL double LoadShapes_Get_SInterval(void);
    DSS_CAPI_V7_DLL void LoadShapes_Set_SInterval(double Value);
```

This is a low priority issue.

PMeira, 2019-02-06:
Yep, that comes from the COM names:
```
    function Get_sInterval: Double; safecall;
    procedure Set_Sinterval(Value: Double); safecall;
```
I'll duplicate them to make them consistent and deprecate one of the pairs.

PMeira, 2019-02-07:
Reopening as Free Pascal doesn't like the approach I used on Linux. The CI didn't fail when it should have, so I'll fix that too.

PMeira, 2019-02-07:
It seems there is a bug in Free Pascal where it makes one of the symbols local when there are two functions with the same name but different casing. It works fine on Windows but breaks on Linux. To work around it, we would have to provide a whole custom linker script. I don't think that's worth it, so I just reverted the change and will rename the functions in 0.11. With that, 0.10.1 should be compatible with 0.10.0.

commit 6e62b119 (2019-02-07): Export consistent casing for LoadShapes_Get_SInterval/LoadShapes_Set_SInterval

Closes #24
PR #26 (2019-02-08): Use custom link options to expose symbol aliases
Different approaches per OS:
- On Windows, listing the same function with different casing exports them both with the casing from the exports block. This is listed as the expected behavior in the FPC docs.
- On Linux, patch the linker scripts to export aliases of the symbols in the global sections, otherwise they get stripped.
- On MacOS, patch the link scripts to use an alias-list, and append the symbols to the export list.

Closes #24
closed 2019-02-08 by PMeira