dss-extensions/dss_capi #29: Moving dss_* and OpenDSSDirect.* projects to a GitHub org

PMeira, 2019-02-09:
With the growing number of alternative interfaces based on DSS C-API, as well as related repositories, the creation of a thematic GitHub organization would facilitate collaboration and sharing of resources:

- GitHub pages with a distinct URL (I could register a dedicated domain if it's available)
- General documentation, including examples and best practices
- Unified discussion of potential future changes of the API and the custom OpenDSS internals
- A dedicated place for smaller/experimental projects

We probably would need to:
1. select a distinct name 
2. create the organization (it would be good to do the same at GitLab, BitBucket and SourceForge)
3. transfer the repositories
4. update docs, general links, maybe the CI instances

Since all short names are probably taken, I'd suggest "DSS" plus a suffix as the org name, e.g.
- DSS-dev: just the basic
- DSS-WG
- DSS-CoG: collaboration/community group? -- would be easy to create a simple logo and other visuals

@kdheepak, @tshort: Any suggestions or general thoughts about this?

kdheepak, 2019-02-11:
Hi @PMeira, this is a great idea and it'll be great to have all the interfaces in one place! 

`dss-dev` is simple and nice but appears to be taken (unless you already reserved it). I'd go with that if it were available. 

If that's not available, I also like `dss-extensions` or `dss-interfaces`. The name `dss-extensions` also works if we want to make other models of controllers / inverters / other components available to the open source community. I think since it is a github "organization", maybe having "group" in the name is somewhat redundant, so I'd go for `dss-wg` or `dss-cog` if we didn't have any other choice.

Naming is hard.

PMeira, 2019-02-11:
@kdheepak I like `dss-extensions` too, it seems to fit quite well. 

Since I can't immediately think of a better name, I just created it and added you as an owner.  Of course we could still choose another name if you can think of anything else.

kdheepak, 2019-02-15:
New organization is here: https://github.com/dss-extensions

closed 2019-02-15 by kdheepak