dss-extensions/dss_capi #31: Document known differences from the official OpenDSS distribution

PMeira, 2019-02-14:
There are some notable differences (such as lack of plots) and lots of tiny changes and fixes not included in the upstream version. Documenting them will help both users and developers.

kdheepak, 2019-02-15:
We should possibly add a benchmarking suite as well? It should be nice if we could make sure we are getting the same results as the official OpenDSS distribution across a wide range of feeders and run those benchmarks on a CI as part of a nightly/weekly/monthly build.

PMeira, 2019-02-16:
I run most of the sample systems to validate with COM in Python, I can try something in AppVeyor targetting the next release. Tracking in https://github.com/dss-extensions/dss_python/issues/9 and https://github.com/dss-extensions/dss_python/issues/10

Benchmarking will be important version 0.11, and need to be extended for other languages. I'm probably adding a comparison between the performance of the offical COM vs the dss_sharp COM too, just to be informative (I don't expect it to be better, but who knows).

commit b2a388a2 (2019-03-25): Initial "Known differences" document

Closes #31
closed 2019-03-25 by PMeira