dss-extensions/dss_capi #35: Is there a way to disable `Show` globally

kdheepak, 2019-02-20:
I have some OpenDSS files that contain lines like:

```
Show Voltage LN Nodes
Show Currents Elem Resid
Show Powers kVA elem
```

These pop up notepad editors when run in OpenDSS. This is a very useful default to have, but I want to disable these in select runs without editing the files. Is there a global setting I can toggle that prevents them from opening up in an editor?

PMeira, 2019-02-20:
Right now, on Linux/MacOS, setting the editor to `/bin/true` or something like that might suffice -- have you tried that? On Windows I'd have to check the code to see if a simple `cmd /c echo` is enough or if we'd need an actual EXE.

For all interactive features, there'll be an extended API in the future, through https://github.com/dss-extensions/dss_python/issues/3

kdheepak, 2019-02-20:
Is it using the `EDITOR` environment variable? ? Because I have my `EDITOR` environment variable set to `vim` everywhere. But it still pops up TextEdit every time. I tried changing it to `/bin/true` interactivity as well.

I think the problem is that it is using `open -t` to open files, and I believe that does not rely on the `EDITOR` environment variable.

[image]

PMeira, 2019-02-20:
The initial value is hardcoded in `DSSGlobals.pas`. Very easy to change use the EDITOR env var though (the font parameters don't have any use):
```pascal
   {$IFDEF Darwin}
      DefaultEditor   := 'open -t';
      DefaultFontSize := 12;
      DefaultFontName := 'Geneva';
   {$ENDIF}
   {$IFDEF Linux}
      DefaultEditor   := 'xdg-open';
      DefaultFontSize := 10;
      DefaultFontName := 'Arial';
   {$ENDIF}
   {$IF (defined(Windows) or defined(MSWindows))}
      DefaultEditor   := 'NotePad.exe';
      DefaultFontSize := 8;
      DefaultFontName := 'MS Sans Serif';
   {$ENDIF}
```
It gets used as 
```pascal
{$IF (defined(Windows) or defined(MSWindows))}
            RunCommand(DefaultEditor, [FileNm], s);
{$ELSE}
            RunCommand('/bin/bash', ['-c', DefaultEditor + ' ' + FileNm], s);
{$ENDIF}
```

The output of the [RunCommand](https://www.freepascal.org/docs-html/fcl/process/runcommand.html) function in Free Pascal is not being checked, so using any invalid EXE works right now. I consider it a bug though, this needs a proper check.

What I meant about setting the editor is using the `set editor=...` command, e.g. 

```
from opendssdirect import dss
dss.run_command('redirect "IEEETestCases/13Bus/IEEE13Nodeckt.dss"')
dss.run_command('set editor="exe_that_doesnt_exist"')
dss.run_command('show voltages')
```

Using a proper executable here that does nothing and ignores the parameters should work in any version, even after the output of `RunCommand` is handled properly in a future version. This also would be compatible with the official DLL.

kdheepak, 2019-02-21:
Awesome! That works.

closed 2019-02-21 by kdheepak

PMeira, 2019-02-24:
@kdheepak Just added the error checks and a [pair of functions](https://github.com/dss-extensions/dss_python/commit/6ea4552fca31548a4a38c91e07fc9931059f8e18#diff-21cec5c75d15d2f687de9e408847d4bc) to control the behavior as a whole.

kdheepak, 2019-02-26:
Awesome!