dss-extensions/dss_capi #42: Merge electricdss-src into dss_capi?

PMeira, 2019-03-05:
Initially DSS C-API was developed in a separate repo to facilitate merging with the official repo -- the idea was to only touch the interface code (hence the name). That's unlikely to happen and the more changes electricdss-src accumulates to handle bug fixes, FPC/multi-platform adjustments, and custom features, the less it makes sense to keep it as an alternative repository. 

Since nowadays most of the upstream OpenDSS development is done directly on SVN trunk, the per commit changes don't end up that useful -- most are partial or are changed after a couple of commits, making it cumbersome to merge (especially since there's `Version7` and `Version8`). Until December, I was mostly cherry-picking commits, but recently I'm usually reformatting the code for comparison and cherry-picking lines instead. Some features I find unfinished/buggy I'm not picking (mostly from `Version8`).

Merging would mean changing the folder structure in dss_capi a bit, but should simplify development as a whole. E.g. https://github.com/dss-extensions/dss_capi/pull/41 and https://github.com/dss-extensions/electricdss-src/pull/10 address the same thing in the two repositories.

@kdheepak @tshort Would you have any objections about it?

kdheepak, 2019-03-05:
This sounds good to me! As long as we are able to get updates from upstream merged in at some regular interval I don't see why not. Whatever makes your life easier @PMeira :)

PMeira, 2019-03-05:
Great! I'll do it before merging #41 then, already done most of the work on merging and everything worked fine. 

The whole project history is now in this repo!

PR #44 (2019-03-05): Merge electricdss-src into dss_capi
Closes #42
closed 2019-03-05 by PMeira