dss-extensions/dss_capi #53: Contact Information for License Question.

yim0331, 2019-04-01:
Hello,

I'm writing on this issue to the contact information for asking the license question with dss_capi and electricdss-src projects. I want to create a Docker image that would contain the two projects for running OpenDSS in ubuntu system and would be open to the public via Docker hub. I wish to discuss the details via email. Could you please let me know your contact information or send an email to yim@anl.gov?  

Thanks,

PMeira, 2019-04-01:
@yim0331 Hi,
I think most information is already listed on the [README](https://github.com/dss-extensions/dss_capi/blob/master/README.md), isn't it? Including my email (I don't list it everywhere to reduce spam). Repeating some of it and adding a bit:

- the electricdss-src repo has merged into this repo, see #42 
- KLUSolve has moved to its [own repo](https://github.com/dss-extensions/klusolve/). The KLUSolve we use is a fork from the upstream version, uses the [LGPL v2+](https://github.com/dss-extensions/klusolve/blob/master/LICENSE). KLUSolve itself uses SuiteSparse, which is LGPL or GPL depending on the components of the library; Eigen3 is also used in the current master branch.
- the relevant licenses for the main code here are https://github.com/dss-extensions/dss_capi/blob/master/OPENDSS_LICENSE and https://github.com/dss-extensions/dss_capi/blob/master/LICENSE
- closed source components from the official OpenDSS distribution from EPRI (like [DSSView.exe](https://sourceforge.net/p/electricdss/code/2581/tree/trunk/Distrib/x64/DSSView.exe), used for plots, or the newer [OpenDSS-G](https://sourceforge.net/projects/dssimpc/) application) are not included, so don't need to worry about them.

I'm not a lawyer so I'm not sure I can help much about the licenses. Since the projects (DSS C-API and KLUSolve) are basically extended forks, I didn't choose the licenses for DSS C-API or the DSS-Extensions fork of KLUSolve, only added my copyright info. For DSS Python and most of the projects I started, I kept the BSD 3-Clause License for consistency. Check each repo for details.

There are some sites like https://tldrlegal.com/ and https://opensource.org/licenses that could provide some basic understanding of the licenses. 

On the Docker aspect, it should be trivial to build an image for Python or Julia, since the distribution already include or download the dependencies on installation (from [here](https://github.com/dss-extensions/dss_capi/releases)). By the same reason, Docker doesn't add that much value (but I understand Docker is required in some contexts).

Note that DSS C-API is distributed as a library and there is no executable right now (no demand for it). The official OpenDSS contains a more limited project to built an executable without the changes of DSS Extensions as [`opendsscmd`](https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version7/Source/CMD/).

yim0331, 2019-04-03:
@PMeira Thank you for the information!
I will use the pre-release objects instead of code on the Docker.
Again, thank you!

PMeira, 2019-04-05:
@yim0331 I'd use the release files (listed on the same page), the pre-releases are just for testing and could be removed later (the tags will remain) -- there are a bunch of pre-releases for 0.10.3 since there were lots of changes. 

The following link should always redirect to the latest release:

https://github.com/dss-extensions/dss_capi/releases/latest

The proper 0.10.3 should be released in a few days.

> BTW Coul you remove this post to hide my email address

Since you already removed the email, let's just close it. Feel free to open other issue tickets (or email me) if you find any issues or have any suggestions to facilitate installation on Docker.

closed 2019-04-05 by PMeira