dss-extensions/dss_capi #54: Inconsistent results between RMatrix and XMatrix values on Windows and Mac

kdheepak, 2019-04-05:
If I run the following OpenDSS circuit,

```
Clear

New Circuit.test_circuit

New Wiredata.wire1 GMR=0.0255000 DIAM=0.7410000 RAC=0.3060000 NormAmps=530.0000 Runits=mi radunits=in gmrunits=ft
New Wiredata.wire2 GMR=0.0044600 DIAM=0.3980000 RAC=1.120000 NormAmps=230.0000 Runits=mi radunits=in gmrunits=ft

New Linegeometry.geometry_1 nconds=4 nphases=3
~ cond=1 Wire=wire1 x=-1.2909 h=13.716 units=m
~ cond=2 Wire=wire1 x=-0.1530096 h=4.1806368 units=m
~ cond=3 Wire=wire1 x=0.5737 h=13.716 units=m
~ cond=4 Wire=wire2 x=0 h=14.648  units=m ! neutral

New Line.Line1 Bus1=bus1.1.2.3 Bus2=bus2.1.2.3
~ Geometry= geometry_1
~ Length=1 units=m

Set Voltagebases=[4.8,34.5,115.0]
Calcvoltagebases
Solve
```

On Windows I get:

```julia
julia> Lines.ZMatrix()
4×4 Array{Complex{Float64},2}:
 0.000244709+0.000882268im   5.81934e-5+0.000344936im    5.7659e-5+0.000469068im   5.76072e-5+0.000481031im
  5.81934e-5+0.000344936im  0.000245786+0.000881161im   5.81934e-5+0.00034525im    5.81408e-5+0.000338483im
   5.7659e-5+0.000469068im   5.81934e-5+0.00034525im   0.000244709+0.000882268im   5.76073e-5+0.000509296im
  5.76072e-5+0.000481031im   5.81408e-5+0.000338483im   5.76073e-5+0.000509296im  0.000740038+0.00101383im
```

On Mac I get:

```julia
julia> Lines.ZMatrix()
4×4 Array{Complex{Float64},2}:
 0.000254461+0.000882268im   5.81934e-5+0.000344936im    5.7659e-5+0.000469068im  5.76072e-5+0.000481031im
  5.81934e-5+0.000344936im  0.000255537+0.000881161im   5.81934e-5+0.00034525im   5.81408e-5+0.000338483im
   5.7659e-5+0.000469068im   5.81934e-5+0.00034525im   0.000254461+0.000882268im  5.76073e-5+0.000509296im
  5.76072e-5+0.000481031im   5.81408e-5+0.000338483im   5.76073e-5+0.000509296im  0.00074955+0.00101383im
```

There appears to be some inconsistency.

kdheepak, 2019-04-05:
If I make the earth model `SIMPLECARSON` or `FULLCARSON`, I get the same result. However, if I make the earth model `DERI` it returns different results on Windows and Mac/Linux.

```
New Line.Line1 Bus1=bus1.1.2.3 Bus2=bus2.1.2.3
~ Geometry= geometry_1
~ Length=1 units=m EarthModel=DERI
```

PMeira, 2019-04-06:
So far, I noted the Bessel I0 and I1 results are slightly off while everything else is identical. Code [here](https://github.com/dss-extensions/dss_capi/blob/b7828e6cf26d4336803506175843332c0399448b/Version7/Source/Shared/mathutil.pas#L471).

The Windows values between C-API and COM match and seem correct (compared to scipy's results, which use ZBESJ from netlib.org). Disabling optimization on Linux didn't change any of the values. 

A couple of years ago I compared the line model from OpenDSS (COM) with my own C++ implementation and there was indeed a small difference due to those values (Bessel I0 and I1), but nothing this pronounced.  Looking now, there seems to be an unused (maybe partial?) port of ZBESJ from Slatec in [MathutilInterface.pas](https://github.com/dss-extensions/dss_capi/blob/b7828e6cf26d4336803506175843332c0399448b/Version7/Source/Shared/MathutilInterface.pas).

I'm going to compare the local variables in the functions to see if there's any clue in them.

kdheepak, 2019-04-06:
Thanks for taking a look! Yes, I didn't mention it here but I checked using the official OpenDSS GUI on Windows and got the same values as the Windows version using the CAPI. It's odd that it only occurs when using `EarthModel=DERI`. I did a few grep searches to check if there was any Windows specific code in the files related to Lines, LineCodes and LineGeometry but didn't find anything that could be the cause of this.

PMeira, 2019-04-06:
Found it:

```pascal
function CMUL(const a, b: complex): complex; inline;
begin
    Result.RE := A.RE * B.RE - A.IM * B.IM;
    Result.IM := A.RE * B.IM + A.IM * B.RE
end;
```
It looks like when the following idiom is used FPC doesn't check for aliasing and overwrites `term.re`  after the first line.
```
term := CMul(zSQR25, term);
```
At first I thought inlining was the issue, but actually the `const` qualifier is the root cause. Since `complex` a record, it's passed by reference when using `const` (it's equivalent to `var` + some checks). From [this old thread](http://lists.freepascal.org/pipermail/fpc-devel/2011-July/024981.html), it seems the issue also could happen with Delphi.

I didn't find a switch to disable the potential `const` optimization. Tomorrow I think I'll just remove any `const` that potentially could cause aliasing. In terms of optimization, since there values are pairs of doubles, I don't think it will impact anything noticeably.

kdheepak, 2019-04-06:
Wow, great job sleuthing! That is a nasty bug, I'm both amazed and terrified. 

Isn't the Windows build compiled using FPC as well? Why didn't the Windows version have the same issue? Surely the Windows FPC compiler is performing the same optimizations as the Mac/Unix FPC compilers.

PMeira, 2019-04-06:
> That is a nasty bug, I'm both amazed and terrified.

Yep, what I don't like (and more people seem to agree even on that long 2011 thread) is that there's no warning from the compiler. It seems like if you want performance with Pascal you need to use C constructs, but that lead to potential bugs and that's when I wonder if there's a reason to keep using it.

> Isn't the Windows build compiled using FPC as well?

It is, but the calling conventions are not the same and that might be what's preventing the issue from manifesting on Windows. I did also try to tweak the compiler flags to see if we could either get it bug out on Windows or remove the bug on Linux.

Good news is that `const` is not used that much in functions that could cause the issue. Besides `Source/Shared/Ucomplex.pas`, there are candidates in `Source/Common/Utilities.pas` and some other sparse files. I'm reviewing the related code and should fix this soon -- after testing, it's a good time to release 0.10.3 before it accumulates too many changes.

PMeira, 2019-04-07:
The tagged 0.10.3a5 (also pre-released) includes the changes related to this (and a few more changes). @kdheepak Can you check it on your side? I'll try to finish the full cross-platform COM comparison before the next release.

The following were changed:

- `Source/Common/Utilities.pas: RestoreCRLF(const S: String): String;`: This was unused, code has been removed (since it was partially broken anyway).
- `Source/Common/MeTIS_Exec.pas: Max(const A, B: Integer): Integer;`: code removed in favor of the `Math` unit.
- `Source/Common/Utilities.pas: ReplaceCRLF(const S: String): String;`: rewritten with StringReplace, it was probably broken on Linux/MacOS before.
- `Source/TCP_IP/TCP_IP.pas: StrippedOfNonAscii(const s: String): String;`: `const` removed. It was probably broken before but we don't use this on DSS C-API (yet?)
- `Source/Shared/Ucomplex.pas: cmul(const a, b: complex): complex;`: `const` removed
- `Source/Shared/Ucomplex.pas: cdiv(const a, b: complex): complex;`: `const` removed

Some functions (e.g. `Bessel_I0` and `Bessel_I1`) could also present issues if the input parameter is the same as the result. For those, I checked it the current usage does it or if it would ever make sense to use it like that.

I also changed `TDSSClass.ClassEdit` and all of its overrides from `function` to `procedure` since the result was always zero and not used anywhere.

Just to confirm, the issue might happen in Delphi too. This discussion on StackOverflow includes some info from the R&D team at Embarcadero: https://stackoverflow.com/a/21032381

This is the list of the functions I checked: 
```
Source/Common/Circuit.pas: SumSelectedRegisters(Const mtrRegisters:TRegisterArray; Regs: pIntegerArray;  count: Integer): Double;
Source/Common/ControlQueue.pas: Push(const Hour: Integer; const Sec: Double; const Code, ProxyHdl: Integer; const Ow ner: TControlElem): Integer; OVERLOAD;
Source/Common/ControlQueue.pas: Push(const Hour: Integer; const Sec: Double; const Code: EControlAction; const Proxy Hdl: Integer; const Owner: TControlElem): Integer; OVERLOAD;
Source/Common/DSSCallBackRoutines.pas: ControlQueuePushCallBack(const Hour: Integer; const Sec: Double; const Code,  ProxyHdl: Integer; Owner: Pointer): Integer; STDCALL;
Source/Common/DSSCallBackStructDef.pas:ControlQueuePush: Function(Const Hour:Integer; Const Sec:Double; Const Code, ProxyHdl :Integer; Owner:Pointer):Integer; StdCall;
Source/Common/ExportResults.pas: TapPosition(const Transformer: TTransfObj; iWind: Integer): Integer;
Source/Common/ShowResults.pas: TapPosition(const Transformer: TTransfObj; iWind: Integer): Integer;
Source/Controls/CapControl.pas: PF1to2(const Spower: Complex): Double;
Source/Controls/RegControl.pas: TRegControlObj.AtLeastOneTap(const ProposedChange: Double; Increment: Double): Doubl e;
Source/General/LoadShape.pas: Set_Result_im(const realpart: Double): Double;
Source/General/Spectrum.pas: TSpectrumObj.GetMult(const h: Double): Complex;
Source/General/TCC_Curve.pas: GetTCCTime(const C_Value: Double): Double;
Source/General/TCC_Curve.pas: GetUVTime(const V_Value: Double): Double;
Source/General/TCC_Curve.pas: GetOVTime(const V_Value: Double): Double;
Source/Meters/EnergyMeter.pas: MyCount_Avg(const Value: Double; const count: Integer): Double;
Source/Meters/Sensor.pas: TSensorObj.RotatePhases(const j: Integer): Integer;
Source/Meters/Sensor.pas: TSensorObj.LimitToPlusMinusOne(const i: Integer): Integer;
Source/Parser/ParserDel.pas: EncloseQuotes(const s: String): String;
Source/Parser/ParserDel.pas: TestEmpty(const s: String): String;
Source/Parser/ParserDel.pas: GetToken(const LineBuffer: String; var LinePos: Integer): String;
Source/PCElements/Equivalent.pas: TEquivalentObj.DoTerminalsDef(const n: Integer): Integer;
Source/PCElements/IndMach012.pas: Sign(const x: Double): Double;
Source/PCElements/Load.pas: InterpolateY95_YLow(const Vmag: Double): Complex;
Source/PCElements/Load.pas: InterpolateY95I_YLow(const Vmag: Double): Complex;
Source/PDElements/AutoTrans.pas: TrapZero(const Value: Double; DefaultValue: Double): Double;
Source/PDElements/AutoTrans.pas: ZeroTapFix(const tapvalue: Double): Double;
Source/PDElements/Fault.pas: Cube(const X: Double): Double;
Source/PDElements/Transformer.pas: TrapZero(const Value: Double; DefaultValue: Double): Double;

Source/Common/Utilities.pas: GetNodeString(const Busname: String): String;
Source/Common/Utilities.pas: Pad(const S: String; Width: Integer): String;
Source/Common/Utilities.pas: PadDots(const S: String; Width: Integer): String;
Source/Common/Utilities.pas: PadTrunc(const S: String; Width: Integer): String;
Source/Common/Utilities.pas: EncloseQuotes(const s: String): String;
Source/Common/Utilities.pas: ConstructElemName(const Param: String): String;
Source/Common/Utilities.pas: CmulReal_im(const a: Complex; const Mult: Double): Complex;
Source/Common/Utilities.pas: QuadSolver(const a, b, c: Double): Double;
Source/Common/Utilities.pas: checkforblanks(const S: String): String;
Source/Common/Utilities.pas: Powerfactor(const S: Complex): Double;
Source/Common/Utilities.pas: ConvertPFToPFRange2(const value: Double): Double;
Source/Common/Utilities.pas: ConvertPFRange2ToPF(const value: Double): Double;
Source/Common/Utilities.pas: StripExtension(const S: String): String;
Source/Common/Utilities.pas: StripClassName(const S: String): String;
Source/Common/Utilities.pas: PFSign(const S: Complex): Double;
Source/Common/Utilities.pas: ExtractComment(const s: String): String;
[... cut ...]
Source/Shared/Ucomplex.pas: ptocomplex(const a: polar): complex;
Source/Shared/Ucomplex.pas: padd(const a, b: polar): polar;
Source/Shared/Ucomplex.pas: psub(const a, b: polar): polar;
Source/Shared/Ucomplex.pas: pmul(const a, b: polar): polar;
Source/Shared/Ucomplex.pas: pdiv(const a, b: polar): polar;
Source/Shared/Ucomplex.pas: pdegtocomplex(const magn, angle: Double): complex;
Source/Shared/Ucomplex.pas: pclx(const magn, angle: Double): complex;
Source/Shared/Ucomplex.pas: cmplx(const a, b: Double): complex;
```

kdheepak, 2019-04-07:
Unfortunately when I use any of the `0.10.3x` releases I get a segfault:

```
julia --color=yes --project
               _
   _       _ _(_)_     |  Documentation: https://docs.julialang.org
  (_)     | (_) (_)    |
   _ _   _| |_  __ _   |  Type "?" for help, "]?" for Pkg help.
  | | | | | | |/ _` |  |
  | | |_| | | | (_| |  |  Version 1.1.0 (2019-01-21)
 _/ |\__'_|_|_|\__'_|  |  Official https://julialang.org/ release
|__/                   |

julia> using OpenDSSDirect

julia> dss("""
       Clear

       New Circuit.test_circuit

       New Wiredata.wire1 GMR=0.0255000 DIAM=0.7410000 RAC=0.3060000 NormAmps=530.0000 Runits=mi radunits=in gmrunits=ft
       New Wiredata.wire2 GMR=0.0044600 DIAM=0.3980000 RAC=1.120000 NormAmps=230.0000 Runits=mi radunits=in gmrunits=ft

       New Linegeometry.geometry_1 nconds=4 nphases=3
       ~ cond=1 Wire=wire1 x=-1.2909 h=13.716 units=m
       ~ cond=2 Wire=wire1 x=-0.1530096 h=4.1806368 units=m
       ~ cond=3 Wire=wire1 x=0.5737 h=13.716 units=m
       ~ cond=4 Wire=wire2 x=0 h=14.648  units=m ! neutral

       New Line.Line1 Bus1=bus1.1.2.3 Bus2=bus2.1.2.3
       ~ Geometry= geometry_1
       ~ Length=1 units=m

       Set Voltagebases=[4.8,34.5,115.0]
       Calcvoltagebases
       Solve

       """)
""

julia> Lines.ZMatrix()

signal (11): Segmentation fault: 11
in expression starting at no file:0
Capacitors_Set_idx at /Users/$USER/GitRepos/OpenDSSDirect.jl/deps/apple/libdss_capi_v7.dylib (unknown line)
Allocations: 9730695 (Pool: 9728350; Big: 2345); GC: 21
[1]    87804 segmentation fault  /Users/$USER/Applications/Julia-1.1.app/Contents/Resources/julia/bin/julia

```

The above was with v0.10.3a5. I've tried compiling locally as well and still get the same segfault.

kdheepak, 2019-04-07:
On Windows oddly, I get the following error when trying to execute the `Calcvoltagebases` line:

```
julia> dss("""Calcvoltagebases""")
ERROR: ReadOnlyMemoryError()
Stacktrace:
 [1] Text_Set_Command(::String) at Z:\GitRepos\OpenDSSDirect.jl\src\lib.jl:4374
 [2] Command(::String) at Z:\GitRepos\OpenDSSDirect.jl\src\text.jl:25
 [3] top-level scope at none:0
```

PMeira, 2019-04-07:
I really got to automate testing with all interfaces to avoid this kind of issue. The crash itself was another (simple) memory corruption I accidentally introduced in 0.10.3 while refactoring. 

Testing that, I found another issue: in COM, `AllNames` for some classes change the active element, others don't. When I started changing these functions a while ago, some classes ended up inconsistent with COM -- that is, `AllNames` left a different element active than the one from COM (when it does change the active element, it's left as the last one of the list). I changed this to be consistent and added a test case. One test needed to be updated in OpenDSSDirect.jl to address that.

I tagged another pre-release and created a branch in OpenDSSDirect.jl to run the test suite (I didn't new tests though). This time it seems to be fine: https://travis-ci.com/dss-extensions/OpenDSSDirect.jl/builds/107382777

I'll merge the upstream changes from SVN and tag another pre-release right after.

kdheepak, 2019-04-07:
Thanks for looking into this! Let me know if I can do anything.

kdheepak, 2019-04-08:
With the latest tag, when I compile locally, I don't get the segfault anymore on my Mac, but I am still getting the old `ZMatrix` values.

```
julia> Lines.ZMatrix()
4×4 Array{Complex{Float64},2}:
 0.000254461+0.000882268im   5.81934e-5+0.000344936im    5.7659e-5+0.000469068im  5.76072e-5+0.000481031im
  5.81934e-5+0.000344936im  0.000255537+0.000881161im   5.81934e-5+0.00034525im   5.81408e-5+0.000338483im
   5.7659e-5+0.000469068im   5.81934e-5+0.00034525im   0.000254461+0.000882268im  5.76073e-5+0.000509296im
  5.76072e-5+0.000481031im   5.81408e-5+0.000338483im   5.76073e-5+0.000509296im  0.00074955+0.00101383im

julia>
```

This is still inconsistent between Windows and Mac.

PMeira, 2019-04-08:
I tested it yesterday on Mac and got the expected result. Can you try the [pre-built binaries](https://github.com/dss-extensions/dss_capi/releases/tag/0.10.3a7)? If those don't work, I think we'll need to test the related code in isolation to understand what's happening.

Yesterday I also merged the upstream changes but had to fix a small issue. I reported it here:
https://sourceforge.net/p/electricdss/discussion/experts/thread/8f91b8dc29/

The merge from upstream might also have introduced a GUI dependency that's breaking console interaction (Windows only). I'll look into it later, unfortunately there's no easy way to automate a test for this.

kdheepak, 2019-04-08:
Interesting. The pre-built binaries give me the expected result, i.e. same result as Windows.

```julia
Lines.ZMatrix()
4×4 Array{Complex{Float64},2}:
 0.000244709+0.000882268im  …   5.76072e-5+0.000481031im
  5.81934e-5+0.000344936im      5.81408e-5+0.000338483im
   5.7659e-5+0.000469068im      5.76073e-5+0.000509296im
  5.76072e-5+0.000481031im     0.000740038+0.00101383im
```

PMeira, 2019-04-08:
Any chance FPC is reusing old object files on Mac?

kdheepak, 2019-04-08:
I did a `git clean -fxd` but I think I copied over the incorrect file to the `OpenDSSDirect.jl/deps/apple` folder. This newest tag resolved the issue for me, if I compile locally and if I download the pre-built binaries!

PMeira, 2019-04-08:
That's great! I was starting to worry. I started using symlinks to avoid copying the test binaries on Windows/Linux (that bit me in the past too).

I'll try to finish the cross-platform validation to see it there are other inconsistencies like that before the proper release.

PMeira, 2019-04-10:
> I'll try to finish the cross-platform validation to see it there are other inconsistencies like that before the proper release.

Finally did that and, fortunately, no surprises, the errors are now in the same order as the Windows build! Basically I ran [`python validate.py save`](https://github.com/dss-extensions/dss_python/blob/more_idx/tests/`) on Windows to pickle the COM output, then used the pickled results on Linux and macOS. I had to rename files to fix the casing for Linux (updated in electricdss-tst).

The saved results compress nicely to around 50 MB -- if it doesn't take too long to run on Travis, we can upload that somewhere and run it when building dss_python; if it does take long, a subset without the large circuits should be fine to get good coverage. I'll follow up on https://github.com/dss-extensions/dss_python/issues/9

Although the process didn't find new bugs, #15 still needs some treatment as the current solution failed to error on a missing file.

I also ran the validation without the fix for this issue and, yes, it caught on the first test circuit. I looked into our files at Unicamp to check why we didn't get the error before and, besides other potential changes in the build, the files we use/generate generally include the LineCodes directly -- the LineGeometries are post-processed during generation and fed back as LineCodes, which are also exported to a DB for other studies without OpenDSS. So far this was done using Windows, we got lucky...

closed 2019-04-10 by PMeira

kdheepak, 2019-04-10:
Nice! Travis-CI lets you run cron jobs, long running tests can be run every night or every week. 

When you compare COM results on Windows is it using dss_capi? Can we run it against the official library as well?

PMeira, 2019-04-10:
> Nice! Travis-CI lets you run cron jobs, long running tests can be run every night or every week.

OK, running nightly and skipping the tests if there are no changes in any of the related repositories looks like a good idea. 

I'll try to rewrite the validation script to use nose, pytest or hypothesis. I started something about it but never finished it.

> When you compare COM results on Windows is it using dss_capi? Can we run it against the official library as well?

The script saves the official COM results (the errors I mentioned were in relation to them) and compares those with dss_python's -- we don't have an actual COM implementation yet (DSS# can expose one with some extra steps).

kdheepak, 2019-04-10:
I'm not sure I exactly follow. I also get a "page not found" when I try to visit the link you posted when I visit from my mobile, I'll try again later and post my question more clearly.

PMeira, 2019-04-10:
Fixed, there was a typo there.