dss-extensions/dss_capi #55: Unable to compile latest releases

kdheepak, 2019-04-06:
If I try to compile `0.10.2` I get the following error:

```
Compiling ./src/v7/CAPI_CmathLib.pas
CAPI_CmathLib.pas(36,12) Warning: range check error while evaluating constants (1 must be between 0 and 0)
CAPI_CmathLib.pas(64,12) Warning: range check error while evaluating constants (1 must be between 0 and 0)
CAPI_CmathLib.pas(82,12) Warning: range check error while evaluating constants (1 must be between 0 and 0)
CAPI_CmathLib.pas(100,12) Warning: range check error while evaluating constants (1 must be between 0 and 0)
CAPI_CmathLib.pas(118,12) Warning: range check error while evaluating constants (1 must be between 0 and 0)
Compiling ./src/v7/CAPI_CNData.pas
Compiling ./src/v7/CAPI_WireData.pas
Compiling ./src/v7/CAPI_CtrlQueue.pas
CAPI_CtrlQueue.pas(139,1) Warning: Function result variable does not seem to initialized
Compiling src/v7/CAPI_DSS.pas
CAPI_DSS.pas(64,15) Error: Identifier not found "DSS_CAPI_ALLOW_EDITOR"
CAPI_DSS.pas(69,5) Error: Identifier not found "DSS_CAPI_ALLOW_EDITOR"
CAPI_DSS.pas(220) Fatal: There were 2 errors compiling module, stopping
Fatal: Compilation aborted
Error: /usr/local/bin/ppcx64 returned an error exitcode
```

It's possible the above error was fixed in the latest tagged releases, I'm not able to finish the compilation process on them either. I believe it finishes compiling/assembling but fails during the linking step.

If I try the same process on any of the `0.10.3ax` releases, I get the following error:

```
e (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/versionresource.o) was built for newer OSX version (10.9) than being linked (10.5)

ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/resdatastream.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/resourcetree.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/resmerger.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/resfactory.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/stringtableresource.o) was built for newer OSX version (10.9) than being linked (10.5)object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/groupiconresource.o) was built for newer OSX version (10.9) than being linked (10.5)

ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/groupcursorresource.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/groupresource.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/fcl-res/icocurtypes.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/regexpr/regexpr.o) was built for newer OSX version (10.9) than being linked (10.5)
ld: warning: object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/rtl-objpas/varutils.o) was built for newer OSX version (10.9) than being linked (10.5)ld: warning:
object file (/usr/local/Cellar/fpc/3.0.4_1/lib/fpc/3.0.4/units/x86_64-darwin/rtl-objpas/variants.o) was built for newer OSX version (10.9) than being linked (10.5)
Undefined symbols for architecture x86_64:
  "_mvmult", referenced from:
      _UCMATRIX$_$TCMATRIX_$__$$_MVMULT$PCOMPLEXARRAY$PCOMPLEXARRAY in Ucmatrix.o
ld: symbol(s) not found for architecture x86_64
An error occurred while linking lib/darwin_x64/libdss_capi_v7.dylib
+ DSS_CAPI_LIB=lib/darwin_x64/libdss_capi_v7.dylib
++ otool -L lib/darwin_x64/libdss_capi_v7.dylib
++ grep libklusolve
++ cut -f 1 -d ' '
++ sed 's/^[    ]*//'
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/objdump: 'lib/darwin_x64/libdss_capi_v7.dylib': No such file or directory
+ CURRENT_LIBKLUSOLVE=
+ NEW_LIBKLUSOLVE=@loader_path/./libklusolve.dylib
+ install_name_tool -change '' @loader_path/./libklusolve.dylib lib/darwin_x64/libdss_capi_v7.dylib
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/install_name_tool: can't open file: lib/darwin_x64/libdss_capi_v7.dylib (No such file or directory)
```

I'm using the following steps on my Mac.

```bash
chmod +x make_metadata.sh
chmod +x build_macos_x64.sh
./make_metadata.sh
./build_macos_x64.sh
```

kdheepak, 2019-04-06:
I had to download the latest `libklusolve.dylib` and it worked.

closed 2019-04-06 by kdheepak

PMeira, 2019-04-06:
Not sure what happened with the first issue (missing `DSS_CAPI_ALLOW_EDITOR`), the tagged 0.10.2 build fine:
https://travis-ci.com/dss-extensions/dss_capi/jobs/181488014

The unreleased 0.10.3 from master is when klusolve was split, that's why it needs the separate download/build.