dss-extensions/dss_capi #60: v7: Process-based parallelism

PMeira, 2019-05-23:
Comparing the performance, past bugs and some remaining issues with the Version 8 codebase compared to Version 7, I can think of two general approaches for porting the actor-based parallel-machine features for Version 7:

1. Encapsulate an OpenDSS instance in an object, removing most of the global state from OpenDSS to make it more thread-safe. This could be a large task, so it might be better to do it in a whole alternative implementation.

2. Use an IPC mechanism and shared memory constructs to simplify actors, while still maintaining most of the features from Version 8 in Version 7.

IPC should be easier to implement, so we can at least give it a try. If it's successful, we can drop our Version 8 implementation and be confident that most of the Version 8 bugs won't be present.

PMeira, 2019-05-23:
[SimpleIPC](https://www.freepascal.org/docs-html/fcl/simpleipc/tsimpleipcserver.html) seems to be included in the default FPC distribution, I'll start with that.

toanngosy, 2020-06-11:
Hi @PMeira, I have 2 processes and on each process, I have an OpenDSS instance. These 2 instances could run independently without blocking each other? Is there any chance that they would wait for each other to use the solver?

PMeira, 2020-06-17:
Hi, @toanngosy ,
Yes, in general. I have an old example here: https://sourceforge.net/p/electricdss/discussion/861976/thread/5703a79b3b/#4dbf

The processes are independent, the only issue would be if you produce, for example, text files (`export ...`, `show ...`) for the same circuit on the same folder. A future version will allow you to set a file-name suffix to make that possible.

A couple of versions of the official OpenDSS 8+ on Windows had some issues but that never affected DSS C-API.

toanngosy, 2020-06-20:
Thank you so much, I will look into it.

PMeira, 2021-04-23:
Closing since a release in the coming future will still use threads, initially. If that doesn't turn out great, we'll get back to the process approach.

closed 2021-04-23 by PMeira