dss-extensions/dss_capi #64: Maximum path length limited to 255 characters

kdheepak, 2019-11-11:
We are running into a weird error on Linux machines when calling `compile /full/path/to/file.dss`, when the path is more than 255 characters. The error returned by OpenDSS is not obvious, since it just says it cannot open `/full/path/to/file.dss` (without truncating).

kdheepak, 2019-11-11:
[image]

kdheepak, 2019-11-11:
I'm assuming this is not a straightforward fix, since OpenDSS is just calling built-in pascal functions: https://github.com/dss-extensions/dss_capi/blob/cf442f6b93a467f4f74b3dc2206598c23f07fcff/Version7/Source/Executive/ExecHelper.pas#L347-L406

PMeira, 2019-11-11:
Weird, FPC's `MAX_PATH` is 4096 on Linux and still fails. `FileExists(ReDirFile)` returns true though. I'll try to look for an alternative to `AssignFile`.

PMeira, 2019-11-11:
What I found so far is that it doesn't seem to be related to string encoding or anything like it. Using a `TFileStream` seems to work, I only worry about the encoding of the contents.

Related but closed for 3.0.0 (we're on 3.0.4): https://bugs.freepascal.org/view.php?id=24885

FPC 3.2 was expected this year. If it's indeed a bug, might be worth report it before the release (assuming it still existing in the development version).

PMeira, 2019-11-11:
I'm considering using a stream as a fallback if `FileExists` returns true -- just load it in a `TStringList` and iterate over the lines.

It will use a bit more memory since it loads all lines at once. I assume that's not an issue.

kdheepak, 2019-11-11:
We might be able to get away with just removing this one line:

https://github.com/dss-extensions/dss_capi/blob/cf442f6b93a467f4f74b3dc2206598c23f07fcff/Version7/Source/Executive/ExecHelper.pas#L369

kdheepak, 2019-11-11:
I can't think of a reason this function would need to `ExpandFileName`. Even if it does, we can add it as a separate `try` `except` and we should be good.

PMeira, 2019-11-12:
Yeah, `ExpandFileName` doesn't look it does anything useful here, weird. There might be some corner case.

Searching further, this thread from 2018 mentions exactly the same error (the issue is with the `Reset` function): https://forum.lazarus.freepascal.org/index.php?topic=39522.0 
I didn't find the exact bug report on https://bugs.freepascal.org/ but the ones I found suggest using `TFileStream` too, `Assign` and `Reset` are deprecated and won't be fixed (!).

Besides the change to `DoRedirect`, I ported some other changes to 0.10.x and created a branch (master is 0.11 dev), if you want to give it a try: https://github.com/dss-extensions/dss_capi/commit/96bc0bb41075b6627c9ad73e929ad04443860004

It tries the following:
- given filename
- expanded filename
- if the expanded filename does exist, fallback to a string list -- won't affect other files
- otherwise, fail

I tested with just a file name and a full long path:
- `'compile 40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05.dss'`
- `compile /home/meira/40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05/40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05-40242748-21fd-4a61-8ad9-cefdd3ff6a05.dss'`

I'll add a couple of minor fixes from upstream tomorrow on that branch, and hopefully merge a monstrous changeset I have locally too on master.

Since OpenDSS uses `SetCurrentDir`, `DoRedirect` is probably broken in some multi-threading scenarios. I'll open a ticket about it.

kdheepak, 2019-11-12:
You are awesome @PMeira! I'll try the fix.

PMeira, 2019-11-12:
@kdheepak I tagged it so the binaries should be here soon https://github.com/dss-extensions/dss_capi/releases

closed 2019-11-16 by PMeira