dss-extensions/dss_capi #69: ARM support

PMeira, 2019-12-01:
Depends on https://github.com/dss-extensions/klusolve/issues/12

Free Pascal has some support for cross-compiling to ARM. This would allow easier ports to Android down the line.

PMeira, 2020-08-07:
The basic compilation works fine for ARM, it doesn't look like there is any issue on the Pascal side of things. Most of the work will be setting up the environments for ARM and ARM64. 

Travis-CI [supports ARM](https://docs.travis-ci.com/user/multi-cpu-architectures/). So, after KLUSolveX is built for ARM architectures, the bulk of the work should be done. 

For macOS, the Free Pascal SVN already can handle ARM64 too, but I didn't test it. That would probably be evaluated later when ARM macs are available for the general public. It could be that the native emulation is enough.

PMeira, 2020-08-11:
I was able to build and run the majority of the DSS_Python validation tests on a 32-bit ARM. I didn't install Pandas to run all ODD.py tests, but the ones that don't require Pandas all ran fine too. For ODD.jl, it's crashing but it's probably not worth investigating since Julia's not really supported anymore on ARMv7 (we can focus on AARCH64).

Sneak peek:

```python
Python 3.6.5 (default, Apr  1 2018, 05:46:30)
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>
>>> import platform
>>> platform.machine()
'armv7l'
>>> from dss import DSS
>>> DSS.AllowEditor = False
>>> DSS.Text.Command = 'compile Run_IEEE34Mod1.dss'
>>> DSS.ActiveCircuit.TotalPower
array([-2030.78789021,  -284.52797797])
>>> DSS.ActiveCircuit.AllNodeNames
['sourcebus.1', 'sourcebus.2', 'sourcebus.3', '800.1', '800.2', '800.3', '802.1', '802.2', '802.3', '806.1', '806.2', '806.3', '808.1', '808.2', '808.3', '810.2', '812.1', '812.2', '812.3', '814.1', '814.2', '814.3', '814r.1', '814r.2', '814r.3', '850.1', '850.2', '850.3', '816.1', '816.2', '816.3', '818.1', '824.1', '824.2', '824.3', '820.1', '822.1', '826.2', '828.1', '828.2', '828.3', '830.1', '830.2', '830.3', '854.1', '854.2', '854.3', '832.1', '832.2', '832.3', '858.1', '858.2', '858.3', '834.1', '834.2', '834.3', '860.1', '860.2', '860.3', '842.1', '842.2', '842.3', '836.1', '836.2', '836.3', '840.1', '840.2', '840.3', '862.1', '862.2', '862.3', '844.1', '844.2', '844.3', '846.1', '846.2', '846.3', '848.1', '848.2', '848.3', '852r.1', '852r.2', '852r.3', '888.1', '888.2', '888.3', '856.2', '852.1', '852.2', '852.3', '864.1', '838.2', '890.1', '890.2', '890.3']
>>>
```

PMeira, 2021-01-12:
ARM64/AARCH64 built and ran fine too.

thaitranp, 2021-04-09:
Hi, will there be an updated build file for arm64 coming soon? I try to compile the code on arm myself but didn't have much success.

Thanks

PMeira, 2021-04-09:
@thaitranp Hi, we can add it for the next release, in a week or two max. Do you need only DSS C-API, or some of the downstream projects too (dss_python, odd.jl, odd.py...)?

Most of the work is to update [klusolve](https://github.com/dss-extensions/klusolve) and prepare binaries there. Here, on the Free Pascal side, besides the actual installation of Free Pascal, the essential change is to copy the x64 scripts/cfg and update this line (and input/output folders):

https://github.com/dss-extensions/dss_capi/blob/0ef72db7eb046ad5d7a3731ff5ea33b8089b728c/src/linux-x64.cfg#L6

The rest is just CI integration and testing, e.g. does the binary built via CI services have good compatibility?

In case something gets in the way, we can make a pre-release/beta first so it doesn't get postponed too much.

I guess 32-bit ARM can wait a bit longer (useful for Android phones with 4GB or less of RAM, or older Raspberry Pi models).

thaitranp, 2021-04-09:
Thank you so much for the quick reply. We need the dss_capi library for our model in dss_python, I'm not sure how much work is involved to get from dss_capi on Arm64 to dss_python on Arm64 but I assume it should be relatively straight forward?

Thanks

PMeira, 2021-04-09:
> I assume it should be relatively straight forward?

Yep! 
Distribution of wheels on PyPI may be an issue but the wheel could be uploaded somewhere else (GitHub, anaconda.org, or SourceForge). 
Worst case scenarios, without pre-built wheels but binaries of dss_capi and klusolve available, you'd need a C compiler to build it. If it comes to that, we can discuss how to make the build instructions more complete for ARM.

thaitranp, 2021-04-09:
awesome! looking forward to your update. Thanks again for all the hard work

PMeira, 2021-04-16:
@thaitranp Here's a preview -- I'll push the changes tomorrow. Be sure to validate some circuits before using it as one of the 116 odd.py tests failed (the results are close even for that one). I'll have to check that and run the full tests we have before releasing.

[dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip](https://github.com/dss-extensions/dss_capi/files/6322996/dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip)

About the wheel inside the zip file: built on a Raspberry Pi 4, with openSUSE 15.2. This version is built with Py_LIMITED_API and the auditwheel tool decided that it's compatible with manylinux2014. In the end, what matters is that it should work on multiple Linux distributions, starting from CPython 3.6 (I tested it under Python 3.8).

thaitranp, 2021-04-16:
thank you very much for the quick release. We will test it on our circuit and let you know if we find bugs/issues

thaitranp, 2021-04-17:
I tried to install the new release today on a pi 3. The installation of the wheel ran without error but when I tried to import dss in python it can't find the lib file. Here's the error
[image]

Thanks

PMeira, 2021-04-18:
@thaitranp Hi, I got an email notification but I can't find your comment here. Did it work? If not, please send the OS version so I can install it and check what's the issue.

thaitranp, 2021-04-18:
Hi sorry about that. I did have some problem with the import in python not seeing the lib file but changing the PATH variable solved it. 

I'm currently still in the process of setting up our code for aarch64 on the Pi (we have been running our project on an x64 machine so far) and haven't been able to fully test dss_python on aarch64 yet. I'll check back and let you know if everything ran smoothly

PMeira, 2021-09-26:
**Update on ARM32** 

The main issue that caused crashes was a pointer-size error in the original KLUSolve, used by the 0.10.x branch. The new version used by the master branch was rewritten and fixed some bugs along the way, so there's no need to work on that -- besides the name change, the new version can be used as-is on Linux.

The only remaining issue I found so far is related to PVSystems, version 9+ version (all other basic tests pass). It's most likely uninitialized variables.

It looks like it won't be that hard to add ARM32 support.

PMeira, 2021-09-27:
Turns out at least one of the ARM bugs/limitations reported in FPC affects us. FPC 3.2.2 (#101) doesn't help here, unfortunately.

In this case, `packed record` seems to corrupt memory (breaks the stack?), both here

https://github.com/dss-extensions/dss_capi/blob/a1be467609314cb4d5b164c3e088591a9fa25567/src/PCElements/PVsystem2.pas#L56

and in the legacy model. 

Removing `packed` makes the basic test pass, no crashes. I imagine more testing would point some more issues related to that, since there are some other uses throughout the code. 

In this case, it might be related to this: https://gitlab.com/freepascal.org/fpc/source/-/issues/30872#note_644664090

> The problem of the given code using format is, that doubles are passed by reference to format and format assumes all references are naturally aligned. This is a common assumption for most libraries that they can assume references are properly aligned. 

`Format` is indeed used, as well as other functions. It doesn't look like there is an easy fixed without removing the usage of packed records.

`packed record` is used mostly to pass data to third-party DLLs (e.g. user-models). We could just drop them for this platform since there is no official OpenDSS for ARM at all, or just drop it completely since our implementation of user-models will probably change down the line anyway. I still haven't seen any public user-model in the wild, so there's probably no harm done. If supporting ancient models becomes a requirement (note that EPRI's version itself doesn't support old models at all), a separate "interface DLL" could be used at the cost of some performance.

PMeira, 2021-09-30:
I'm moving the migration to "KLUSolveX" (the rewritten KLUSolve) to version 0.12 instead of 0.13 so ARM32 is proper supported. On the `packed record` issue, I'm adding a conditional to disable it only on ARM32, since it seems to work fine for other processors/platforms.