dss-extensions/dss_capi #78: `Lines.RMatrix`, `Lines.XMatrix` and `Lines.CMatrix` don't get correct value when using line geometry

kdheepak, 2020-05-22:
Let's say I have the following file:

```opendss
Clear

new circuit.TestFeeder
~ basekv=4.16 pu=1.0000 phases=3 bus1=SourceBus
~ Angle=0
~ MVAsc3=200000 MVASC1=200000

New Line.sourcedestination Units=km Length=1 Bus1=SourceBus.1.2.3 Bus2=DestinationBus.1.2.3 phases=3

Solve
```

If I change `Length=1` to `Length=0.5`, I get the same value for `RMatrix`. This is expected, since `RMatrix` returns the rmatrix per unit length.

```julia
julia> dss("Redirect ./test.dss"); # length = 1

julia> Lines.RMatrix()
3×3 Array{Float64,2}:
 0.0981333  0.0401333  0.0401333
 0.0401333  0.0981333  0.0401333
 0.0401333  0.0401333  0.0981333

julia> Properties.Value("rmatrix")
"[0.09813333 |0.04013333 0.09813333 |0.04013333 0.04013333 0.09813333 ]"

julia> dss("Redirect ./test.dss"); # length = 0.5

julia> Lines.RMatrix()
3×3 Array{Float64,2}:
 0.0981333  0.0401333  0.0401333
 0.0401333  0.0981333  0.0401333
 0.0401333  0.0401333  0.0981333

julia> Properties.Value("rmatrix")
"[0.09813333 |0.04013333 0.09813333 |0.04013333 0.04013333 0.09813333 ]"
```

However, if I have the following file:

```opendss
Clear

new circuit.TestFeeder
~ basekv=4.16 pu=1.0000 phases=3 bus1=SourceBus
~ Angle=0
~ MVAsc3=200000 MVASC1=200000

New WireData.Wire_336.4_aac_oh GMRac=0.006400800000000001 GMRunits=m Diam=0.0169164 Radunits=m normamps=515.0 emergamps=595.0 Rac=0.00020666857220972575 Runits=m

New LineGeometry.Geometry_1 Nconds=4 Nphases=3 Units=ft Cond=1 Wire=Wire_336.4_aac_oh X=0.0 H=38.9 Normamps=515.0 Emergamps=595.0 Cond=2 Wire=Wire_336.4_aac_oh X=3.5 H=38.9 Normamps=515.0 Emergamps=595.0 Cond=3 Wire=Wire_336.4_aac_oh X=10.0 H=38.9 Normamps=515.0 Emergamps=595.0 Cond=4 Wire=Wire_336.4_aac_oh X=5.2 H=35.0 Normamps=515.0 Emergamps=595.0 Reduce=y

New Line.sourcedestination Units=km Length=0.5 Bus1=SourceBus.1.2.3 Bus2=DestinationBus.1.2.3 phases=3 geometry=Geometry_1

Solve
```

And if I change `Length=1` to `Length=0.5`, I get the following:

```julia
julia> dss("Redirect ./test.dss"); # length = 1

julia> Lines.RMatrix()
3×3 Array{Float64,2}:
 0.269308   0.0686486  0.0664062
 0.0686486  0.274694   0.0689824
 0.0664062  0.0689824  0.269908

julia> Properties.Value("rmatrix")
"[0.2693081 |0.06864857 0.2746939 |0.06640624 0.06898236 0.2699083 ]"

julia> dss("Redirect ./test.dss"); # length = 0.5

julia> Lines.RMatrix()
3×3 Array{Float64,2}:
 0.134654   0.0343243  0.0332031
 0.0343243  0.137347   0.0344912
 0.0332031  0.0344912  0.134954

julia> Properties.Value("rmatrix")
"[0.2693081 |0.06864857 0.2746939 |0.06640624 0.06898236 0.2699083 ]"
```

I'm not sure why the `Version` is returning unknown but I'm using the latest tagged version of OpenDSSDirect.jl.

```julia
julia> Basic.Version()
"DSS C-API Library version DEV revision UNKNOWN based on OpenDSS SVN UNKNOWN (v7/classic variation) MVMULT (64-bit build);
License Status: Open "
```

PMeira, 2020-06-23:
The same issue happens in the official version, it's one more inconsistency (probably bug) in the Lines interface. I think the best would be to document all of the issues with the line interface, report to EPRI, and hope it's addressed there first (I can probably provide a patch too). 

Otherwise, I see a couple of less desirable scenarios:
1. Create a new Lines interface to fix most of the issues we saw so far. Use this new interface by default, but add a switch to toggle the official behavior; or
2. Keep as is, but document every inconsistency. Not very fun from the user's point of view

> I'm not sure why the Version is returning unknown but I'm using the latest tagged version of OpenDSSDirect.jl.

Could you test if that's fixed in https://github.com/dss-extensions/OpenDSSDirect.jl/pull/60?

PMeira, 2021-08-15:
Almost there: https://sourceforge.net/p/electricdss/code/3175/ 🙂

PMeira, 2023-03-29:
> Here, RMatrix is not per unit length but the actual RMatrix. This results in subtle hard to identify bugs.

Nowadays, when using LineGeometry, both lines return the same matrix. 

Still, some of the behavior of the line impedance properties can be somewhat confusing. I think adding either adding an option/property to return the effective impedance or to convert the all lines may be useful for some analysis. This would remove a lot of work for users that just want to export the impedances to use in another method/software. Currently `Yprim` could be used but it's not exactly the same.