dss-extensions/dss_capi #80: Add `-g` flag to `fpc` compilation step

kdheepak, 2020-07-08:
Some users are expressing segfaults and want to view the coredump files to see what is going on. Can we make the next release and all releases going forward with the debug enabled?

https://www.freepascal.org/docs-html/current/user/userse53.html

daniel-thom, 2020-07-08:
Because OpenDSS frequently segfaults when there is bad input data this would greatly speed up debug efforts.  We currently have to sprinkle breakpoints or print statements throughout our scripts to identify the problematic lines.  (We use OpenDSSDirect.py.)

PMeira, 2020-07-10:
Since just adding the debug info to a fully optimized build is usually not very useful, we could provide two separate builds. Those could coexist in the same package, for Python, Julia, MATLAB, etc. I suppose the easiest way to switch between them would be an environment variable (maybe DSS_CAPI_DEBUG), which would cause the code to load the target library with minimal code changes.

I think distributing the debug build for the Pascal code is enough for now, no need to worry about KLUSolve(X).

I do already have .cfgs (the files with compiler flags) for Windows and Linux for debug builds, I can add those to the repo. 

If nothing major shows up, I can try to merge some of the recent changes, add the debug builds and prepare a new release in a couple of days.

@kdheepak Do you have any insight on #79? I'm leaning towards adding a new (runtime) compatibility flag.

> segfaults when there is bad input data this

I known some components are more prone to those, e.g. `PVSystem` family (both old and new models). The official OpenDSS even crashes on some of the test files. I think my team doesn't use those much, so we don't find those too frequently, otherwise I'd have already handled them.

@daniel-thom Whenever it's an issue you can reproduce with minimal or shareable data, feel free to post here or email me. The DSS command `Obfuscate` could be useful sometimes.

daniel-thom, 2020-07-10:
@PMeira Here are two examples that I've seen people experience recently:
1. Write code that assumes that PVSystems are present.  Don't handle the case where `dss.PVsystems.First()` returns 0.  Instead, call `dss.Properties.Value(<some-property>)`.  OpenDSS will segfault.  This is obviously user error, but many people I've observed developing scripts with OpenDSSDirect are not thorough with respect to unexpected conditions.  I'm not familiar enough with the architecture to comment on an appropriate result here (exception thrown, invalid value returned, etc).  An error message printed to stderr would be great.
2. Add new elements to a circuit and call `solve`.  Add more elements but do not call `solve`.  Try to read values from any element.  OpenDSS will segfault.  @kdheepak actually debugged this for me a few months ago.  He said that some internal memory will be uninitialized until `solve` is called.  I think that this is a very common problem.  It is the first thing I check when I start to help someone.

PMeira, 2020-07-12:
@daniel-thom Thanks, I can try to add some more checks on the API side of things to avoid some of those issues. More profound changes will come in the future 0.11.x series.

@kdheepak It seems the packaging is working: https://github.com/dss-extensions/dss_capi/releases/tag/0.10.6a1
The debug binary should have a `d` suffix.

I'll work the other issues marked as [0.10.6](https://github.com/dss-extensions/dss_capi/milestone/5) before testing downstream (ODD.jl, dss_python) and closing here.

kdheepak, 2020-07-12:
Great! Thanks for the quick response on this! 

Are you planning to make distribute both release and debug builds then? We can probably check if some environment variable like `ODD_DEBUG=true` exists and load the debug build accordingly.

PMeira, 2020-07-12:
> Are you planning to make distribute both release and debug builds then?

@kdheepak Yes, distributing the both libraries in the same package for convenience. Even if the C++ part (matrix solver) is still optimized, the debug build is around 20-25% slower. I left the optimization level for the debug builds at `-O1` ("quick and debugger friendly").

> We can probably check if some environment variable like `ODD_DEBUG=true` exists and load the debug build accordingly.

Yes, I think an env var is the more generic way (see my first reply), especially since DSS C-API itself can already use some. 

On the implementation side:
- **ODD.jl:** it's trivial to just set a different `LIBRARY` in the `OpenDSSDirect.jl` file.
- **DSS MATLAB:** right now it's hardcoded but we can do the same as ODD.jl, so it's trivial too.
- **Python:** we use CFFI in [(out-of-line) API mode](https://cffi.readthedocs.io/en/latest/overview.html#purely-for-performance-api-level-out-of-line) both for performance and reliability reasons. That means building two CFFI modules. We can then decide which one to load and bind in DSS Python once, or duplicate the logic and also decide that on ODD.py.
- **DSS Sharp:** it probably makes sense to provide debug and release binaries, using matching DSS C-API builds. Same would for the future C++ library, if a header-only version is not enough.

I think using a single variable `DSS_EXTENSIONS_DEBUG` for the dynamic languages makes sense and could simplify documentation. If we make it project specific, we'd need 4 variables: `ODDJL_DEBUG`, `ODDPY_DEBUG`, `DSS_PYTHON_DEBUG`, `DSS_MATLAB_DEBUG`.

PMeira, 2020-07-26:
I merged a lot of changes in ac460b047128b010716305429a17736dfb703c67 that address many of issues we've seen so far. After that, there were a couple of minor issues that I just fixed.

One of the changes introduces an extended error mechanism, enabled by default:

https://github.com/dss-extensions/dss_capi/blob/ac460b047128b010716305429a17736dfb703c67/include/v7/dss_capi.h#L6099-L6117

To implement this, I refactored most of the API code to do the checks in a dedicated function, and return early in each of functions (I've started doing that before, but never addressed all components). E.g. for transformers:

https://github.com/dss-extensions/dss_capi/blob/ac460b047128b010716305429a17736dfb703c67/src/v7/CAPI_Transformers.pas#L80-L98

The others are similar.  They all check the circuit:

https://github.com/dss-extensions/dss_capi/blob/ac460b047128b010716305429a17736dfb703c67/src/CAPI_Utils.pas#L104-L116

If the extended error mechanism is enabled (boolean `DSS_CAPI_EXT_ERRORS`) and invalid state is found, the error interface is used to pass that info. With the mechanism is disabled, the invalid state is ignored as before, so it should be very compatible with previous usage.  Besides the API, the mechanism can also be disabled using an environment variable `DSS_CAPI_EXT_ERRORS=0`.

Since I also added some more pointer checks, this should address some of the segmentation faults, but only those in the API code. I decided not to waste much time in the internal source code since it would make it harder for me to merge the series of changes from 0.11.

As a note, if these checks are somehow slowing things down in some scenario, replacing the `InvalidCircuit` function with a constant `True` (or a `#define ...`), as well as `DSS_CAPI_EXT_ERRORS` to `False`, should be enough to disable most checks.

Some minor tasks before the release:
- [ ] Decide on #79 ("removal" of the old PVSystem/etc.) -- if it's too much work, we can keep it for this release and make another release to address in a week or two.
- [ ] Update change log
- Downstream projects:
    - [ ] Expose the new PDElements functions and ExtendedErrors setting; add some related tests.
    - [ ] Propagate the API changes from DSS Python to ODD.py (mostly automated), including checking for errors in all API/lib calls. Same for ODD.jl.
    - [ ] Test on all platforms (mostly using CI)

In the interim, I found some issues that I reported in the official forum. With that, most of them were addressed in the official OpenDSS too.

PMeira, 2020-08-01:
DSS C-API and DSS Python released: https://github.com/dss-extensions/dss_python/releases/tag/0.10.6

ODD.py is mostly ready at https://github.com/dss-extensions/OpenDSSDirect.py/pull/85

ODD.jl still needs some small changes at https://github.com/dss-extensions/OpenDSSDirect.jl/pull/61

> (...) Add more elements but do not call solve. Try to read values from any element. OpenDSS will segfault. @kdheepak actually debugged this for me a few months ago. He said that some internal memory will be uninitialized until solve is called.

@daniel-thom I added some checks especially in the CktElement interface, but I'm pretty sure a DSS command with bad data can also cause that, so we'll need to add checks in the engine code later to fully handle it. https://github.com/dss-extensions/dss_capi/issues/56

Some methods can throw access violation exceptions, without crashing the whole process. That's still bad since not all invalid uses are caught in invalid memory regions, some just corrupt another variable's content.

closed 2020-08-01 by PMeira