dss-extensions/dss_capi #81: Add `AllLineNames`, `AllLineIMag`, `AllLineIPu` and similar functions

kdheepak, 2020-07-08:
Currently we have `AllBusNames` and `AllBusVMag` in `Circuit` but no `AllLineNames` and `AllLineIMag`. Users are using `export currents` to get a csv file and read the data back into Python / Julia. Maybe this is an upstream feature request, since I imagine this is not trivial to fix.

PMeira, 2020-07-10:
Should not be too complex, we can add it. For a future release/docs, it's important to mark those as extensions though.

PMeira, 2020-07-12:
Looking into the `export` and `show` commands, they include all PDElements. The currents can be in rectangular, polar and sequence forms/components. Some reports use `NormAmps` and `EmergAmps` (capacity reports).

Some could be implemented at the user level, but I agree that they're used enough that we should expose them directly, especially since not everything can be easily done with the current API and it could be error prone.

To simplify the functions names, I propose adding the new functions in the existing `PDElements` module, which is currently very limited.

Tentative list:
- `AllNames`
- `AllCurrents`
- `AllCurrentsMagAng`
- `AllSeqCurrents`
- `AllPowers`
- `AllSeqPowers`
- `AllCapacityNorm` / `AllCapacityEmerg`, or a single `AllCapacity` with both values
- `AllNumPhases`? Or maybe `AllNumNodes`, need to consider the terminals too, etc.

Naming not final but I tried to follow the convention from CktElement, which already contain most functions for individual elements, with the exception of capacity functions.

kdheepak, 2020-07-12:
Following the naming convention from the other modules seems great!

> AllNumPhases? Or maybe AllNumNodes, need to consider the terminals too, etc.

If we have a single phase Line is connected between a 3 phase bus and 2 phase bus, i.e. something like this:

[image]

What would `AllNumPhases` / `AllNumNodes` return?

I think we should set ourselves up to build graph structures to represent the circuit when the class based rewrite happens.

PMeira, 2020-07-13:
For your example, 2 phases times 2 terminals = 4 nodes total. For the current `CktElement.Currents`, 

The issue is that both the number of nodes and terminals vary. Maybe listing the total number of nodes is enough, since most elements would be lines (2 terminals). For the exceptions, like transformers with 3 windings, it's probably easier to get the individual information from other functions.

I'll think a bit more about it. Some extra functions, like `AllMaxCurrents`, would be both useful for the users and easier to interpret.

> I think we should set ourselves up to build graph structures to represent the circuit when the class based rewrite happens.

I have to finish 0.11 first (and 0.11 already contains some useful extras), and after that we should start planning a better API as a whole. As long as the official OpenDSS doesn't change drastically, we can keep the current API based on the active element concept for compatibility, while working on a better API (and better OpenDSS engine as whole).

PMeira, 2020-07-26:
This is done, just need to add tests to dss_python or ODD.py before the next release.

kdheepak, 2020-07-26:
Awesome!

PMeira, 2020-07-30:
Relevant tests in https://github.com/dss-extensions/OpenDSSDirect.py/pull/85

closed 2020-07-30 by PMeira