dss-extensions/dss_capi #86: Header docs: add more metadata

PMeira, 2020-08-03:
Soon it will be interesting to add more metadata to the header docs, in a standardized way.

The idea is to complement the header with enough information to generate APIs for complex numbers and arrays, enumerations, etc., as well as document things better and easier.

We do not have enough bandwidth to handle a dedicated/detail documentation for each of the projects DSS Extensions. With this, and a generic way to translate the metadata for each language-specific API, a good general documentation could be achieve. The general docs could be consumed to provide e.g. docstrings, and each project could extend the docs with their own extensions based on the extra features and/or limitations of the API.

Related: https://github.com/dss-extensions/dss_python/issues/28, https://github.com/dss-extensions/dss_python/issues/4, https://github.com/dss-extensions/OpenDSSDirect.py/issues/80

PMeira, 2020-08-18:
The plan is to use Doxygen + [Breathe](https://breathe.readthedocs.io/en/latest/) to integrate Doxygen with Sphinx. At the Sphinx side, we can use filter extensions to process the header docs to complement with some specific notes per language binding. E.g. for Python, mention if the result is a NumPy array (and which dtype) or a plain list.

With this, most of the API documentation would be available for most languages without duplication of efforts. For some, if not all, programming languages exposed in the DSS Extensions org, we can use the final processed docs for each function to generate the language-specific docs. For Python, Julia, MATLAB, and probably C# (dss_sharp is pending a rewrite), this is feasible. The Sphinx Tabs extension could prove useful: https://github.com/executablebooks/sphinx-tabs (as well as some other [third-party extensions](https://www.sphinx-doc.org/en/master/usage/extensions/index.html#third-party-extensions))

Besides the common API, some other important pieces:
- A general documentation describing some aspects of OpenDSS not covered by the official documentation. For example, the notion of active elements, how the distinct interfaces interact, etc.
- Properly formatted versions of the docs in https://github.com/dss-extensions/dss_capi/tree/master/docs
- Language/project-specific features, such as the binding conventions and so on. These could be handled for each individual package. For example, https://github.com/dss-extensions/OpenDSSDirect.py/issues/86 illustrates that this is desirable for new users.

Regarding translation/localization of the docs, we can try Transifex. This would be easy way to contribute to DSS Extensions. [The requirements are minimal](https://docs.transifex.com/projects/open-source-project). Since Brazil and the US are the countries with the most OpenDSS downloads, Portuguese (BR) would be the priority (and I can handle that).

From my side, I believe I'll be able to dedicate some days to this whole endeavor in September, especially after finishing the 0.11 codebase refresh. With an initial working version, the rest can be complemented and fine tuned as time goes.

PMeira, 2020-12-29:
As a first step, an extract of all commands and properties on the DSS language side is now available at https://github.com/dss-extensions/dss_capi/blob/0.10.x/docs/dss_properties.md

Hopefully we'll see some more progress in January 2021.

PMeira, 2021-01-18:
Reviewing some data and thought of another important thing to add: document when (which version or revision) the property was added in COM and C-API, as well as potential changes in behavior.

mauricioborges, 2022-01-14:
@PMeira I'd like to help on documenting it if you wish. I'm working with the library and I don't have much opendss knowledge so I thought would be productive to do so while I implement things. Do you already have any example here of the documentation standard that you would like to follow?