dss-extensions/dss_capi #92: There is no fpc.exe in linux enviroment. So I need to find pc.1.gz? Or other document? And add its containing folder.

ChengweiLou, 2021-03-09:
Add the folder containing the compiler (fpc.exe) to your PATH environment variable. 
(base) .....:~$ whereis pc
pc: /usr/bin/pc /usr/share/man/man1/pc.1.gz

PMeira, 2021-03-09:
Most Linux distributions include Free Pascal in their packaging system.

Alternatively, you can download from https://www.freepascal.org/download.html 
There are some detailed instructions in https://wiki.freepascal.org/Category:Install

ChengweiLou, 2021-03-10:
> Most Linux distributions include Free Pascal in their packaging systems.
> 
> Alternatively, you can download from https://www.freepascal.org/download.html
> There are some detailed instructions in https://wiki.freepascal.org/Category:Install

Sorry for bothering you again.
So do I need to add the PATH environment variable with ubuntu?

PMeira, 2021-03-10:
If you use apt to install FPC 3.0.4 or the [debs from Lazarus](https://sourceforge.net/projects/lazarus/files/Lazarus%20Linux%20amd64%20DEB/Lazarus%202.0.12/), they're already in the default path.

If you install from the pure FPC binary, you might need to add it to your PATH.