dss-extensions/dss_capi #97: Fix custom_link.sh to work with fpc 3.2.2

beroset, 2021-08-07:
As of fpc 3.2.2, external link scripts are no longer named `link.res` but `link{pid}.res` where the {pid} is substituted with the process ID of the running instance of fpc.  This means that `custom_link.sh` no longer works because it assumes the old name.

PMeira, 2021-08-08:
Switching FPC versions require some extensive testing in all platforms. But, since we probably can drop `custom_link.sh` issue for the planned v0.13, it's better to just wait for that before upgrading. I doubt dropping it now would cause any issues, but just to be sure, it's better to follow the plan to avoid surprises.

As a side note, I know that FPC 3.2.2 is important due to native mac M1 support, but the number of mac users in general is small, (hence M1 users even smaller).

PMeira, 2021-08-08:
@beroset Not sure why you're building the master branch, but I'd suggest sticking to the 0.10.x branch unless you plan to contribute something major (if so, please communicate to keep coordination). For 0.10.x and earlier, there are binaries on the [Releases](https://github.com/dss-extensions/dss_capi/releases) page.

PMeira, 2021-08-30:
Closing in favor of #101 and #102, which are more explicit.

closed 2021-08-30 by PMeira

beroset, 2021-09-01:
You may wish to add instructions for contributors to make it clear how you'd prefer that to be done.  See https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/setting-guidelines-for-repository-contributors

beroset, 2021-09-08:
I understand entirely and thank you for your efforts with this valuable project!  My interest in FPC 3.2 is not for Apple M1 but rather because it can target WebAssembly.  Perhaps I'll see about contributing unit testing to the base EPRI project.

PMeira, 2021-10-04:
> but rather because it can target WebAssembly.

I tried WebAssembly with FPC a while ago but it was very incomplete. Not sure about FPC 3.2.2, but from the roadmap
at https://wiki.freepascal.org/WebAssembly/Roadmap it seems that exceptions might be one of the major features missing. 

If there are issues with file handling, the changes from https://github.com/dss-extensions/dss_capi/commit/458f5c7dc6a19c5cadc51f6ab25a39673087d2d6 show that we can wrap everything and expect it to work (see also #96).

Besides that, I imagine linking with KLUSolve(X) might need some work, but C and C++ support for WebAssembly is more complete/mature.

> Perhaps I'll see about contributing unit testing to the base EPRI project.

That'd be great. A basic testing suite would catch a lot of minor and some major issues. Right now, I don't think there is anything available publicly from EPRI. I know some other things (e.g. OpenDSS-G, OpenDSS Viewer, OpenDSS-GIS) are closed source, so I imagine they could have something internal. From some messages in the forum, it seems the bug tracker is also internal.