dss-extensions/dss_capi #100: Illigal parameter error: -vm6058 while running bash build_linux_arm64.sh in RaspberryPi

fhafi, 2021-08-25:


fhafi, 2021-08-25:
I have installed fpc 3.0.2 in a RaspberryPi. I am trying to install dss_capi. But this issue is occurring while running @fpc @src/linux-arm64.cfg -B src/dss_capi.lpr in the 'build_linux_arm64.sh' file.

PMeira, 2021-08-25:
Try FPC 3.2.0 (exactly) with the 0.10.x branch of DSS C-API.
But note that it's not fully tested yet on ARM64, there may be some issues. There are some potential compiler problems open for ARM in the FPC bug tracker that span all recent FPC versions.

There are also pre-built binaries for Python in https://github.com/dss-extensions/dss_capi/issues/69#issuecomment-820954283
Especially for the Pascal part, you can extract that file (`dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip`) twice (`whl` is also a zip container) to get the binaries, e.g.:

```shell
/tmp> mkdir rpi
/tmp> cd rpi
/tmp/rpi> wget https://github.com/dss-extensions/dss_capi/files/6322996/dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip
Connecting to github-repository-files.githubusercontent.com (github-repository-files.githubusercontent.com)|2606:50c0:8000::154|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7424985 (7.1M) [application/x-zip-compressed]
Saving to: ‘dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip’

dss_python-0.10.8.dev1-cp36-abi3-many 100%[========================================================================>]   7.08M  43.8MB/s    in 0.2s    

2021-08-25 13:11:13 (43.8 MB/s) - ‘dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip’ saved [7424985/7424985]

/tmp/rpi> unzip dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip 
Archive:  dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl.zip
  inflating: dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl  
/tmp/rpi> unzip dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl
Archive:  dss_python-0.10.8.dev1-cp36-abi3-manylinux2014_aarch64.whl
  inflating: dss_python-0.10.8.dev1.dist-info/WHEEL  
  inflating: dss_python-0.10.8.dev1.dist-info/LICENSE  
  inflating: dss_python-0.10.8.dev1.dist-info/RECORD  
  inflating: dss_python-0.10.8.dev1.dist-info/top_level.txt  
  inflating: dss_python-0.10.8.dev1.dist-info/METADATA  
  inflating: dss/_dss_capi_v7.abi3.so  
  inflating: dss/libklusolve.so      
  inflating: dss/v7.py               
  inflating: dss/_dss_GenUserModel.abi3.so  
  inflating: dss/v8.py               
  inflating: dss/enums.py            
  inflating: dss/_dss_StoreUserModel.abi3.so  
  inflating: dss/_dss_PVSystemUserModel.abi3.so  
  inflating: dss/libdss_capi_v7.so   
  inflating: dss/patch_dss_com.py    
  inflating: dss/_dss_CapUserControl.abi3.so  
  inflating: dss/_dss_StoreDynaModel.abi3.so  
  inflating: dss/__init__.py         
  inflating: dss/_dss_capi_v7d.abi3.so  
  inflating: dss/_cffi_api_util.py   
  inflating: dss/libdss_capi_v7d.so  
  inflating: dss/dss_capi_ir/IReduceCkt.py  
  inflating: dss/dss_capi_ir/ICapacitors.py  
  inflating: dss/dss_capi_ir/ILoads.py  
  inflating: dss/dss_capi_ir/IPDElements.py  
  inflating: dss/dss_capi_ir/IDSSElement.py  
  inflating: dss/dss_capi_ir/IXYCurves.py  
  inflating: dss/dss_capi_ir/ILineGeometries.py  
  inflating: dss/dss_capi_ir/IRelays.py  
  inflating: dss/dss_capi_ir/ILineCodes.py  
  inflating: dss/dss_capi_ir/ICtrlQueue.py  
  inflating: dss/dss_capi_ir/IParser.py  
  inflating: dss/dss_capi_ir/IDSSProperty.py  
  inflating: dss/dss_capi_ir/IFuses.py  
  inflating: dss/dss_capi_ir/IYMatrix.py  
  inflating: dss/dss_capi_ir/ICmathLib.py  
  inflating: dss/dss_capi_ir/ITSData.py  
  inflating: dss/dss_capi_ir/ICircuit.py  
  inflating: dss/dss_capi_ir/ILineSpacings.py  
  inflating: dss/dss_capi_ir/IGenerators.py  
  inflating: dss/dss_capi_ir/IBus.py  
  inflating: dss/dss_capi_ir/IDSS_Executive.py  
  inflating: dss/dss_capi_ir/ITopology.py  
  inflating: dss/dss_capi_ir/IReactors.py  
  inflating: dss/dss_capi_ir/ISwtControls.py  
  inflating: dss/dss_capi_ir/IDSSEvents.py  
[... 67 lines cut ...]
  inflating: dss/dss_capi_gr/IError.py  
  inflating: dss/dss_capi_gr/IMeters.py  
  inflating: dss/dss_capi_gr/IVsources.py  
  inflating: dss/dss_capi_gr/ICNData.py  
  inflating: dss/dss_capi_gr/ICktElement.py  
  inflating: dss/dss_capi_gr/ISolution.py  
  inflating: dss/dss_capi_gr/IDSS.py  
  inflating: dss/dss_capi_gr/IActiveClass.py  
  inflating: dss/dss_capi_gr/ISensors.py  
  inflating: dss/dss_capi_gr/IReclosers.py  
  inflating: dss/dss_capi_gr/IISources.py  
  inflating: dss/include/stdint_compat.h  
  inflating: dss/include/dss_StoreDynaModel.h  
  inflating: dss/include/dss_StoreUserModel.h  
  inflating: dss/include/dss_CapUserControl.h  
  inflating: dss/include/dss_GenUserModel.h  
  inflating: dss/include/dss_UserModels.h  
  inflating: dss/include/dss_PVSystemUserModel.h  
  inflating: dss/include/v7/dss_capi.h  
  inflating: dss/include/v8/dss_capi.h  
/tmp/rpi> ls dss/lib*.so -l
-rwxr-xr-x 1 meira users 15212112 Apr 16 06:22 dss/libdss_capi_v7d.so
-rwxr-xr-x 1 meira users  5304848 Apr 16 06:22 dss/libdss_capi_v7.so
-rwxr-xr-x 1 meira users  1140264 Apr 16 06:22 dss/libklusolve.so
```

PMeira, 2021-08-30:
Well, I'm closing this since at this point we get back to the main ticket on https://github.com/dss-extensions/dss_capi/issues/69 -- @fhafi, please feel free to add comments there if you find issues on ARM64.

I added a note there mentioned that ARM32 is still not working yet. I'm getting a second RPi to keep as a dedicated 32-bit test/build bot system, so we can expect that someday we'll have both ARM32 and ARM64 support.

closed 2021-08-30 by PMeira