dss-extensions/dss_capi #111: Error: Illegal Parameter:  -vm6058T

keegit, 2022-06-23:
I am trying to build the C API. I have no prior experience with OpenDSS, or Pascal. The steps I have taken so far to do so are:

1. Installed the free pascal compiler 
 ```
sudo apt install fp-compiler-3.0.4
```
 
2. Downloaded and built `klusolve` from root of `dss_capi` 
```
git clone git@github.com:dss-extensions/klusolve.git`
cmake -DUSE_SYSTEM_SUITESPARSE=OFF -DUSE_SYSTEM_EIGEN3=OFF -DKLUSOLVE_LIB_TYPE=STATIC .. 
cmake --build . --config Release
```

3. Clone the `dss-extensions/dss_capi` repository into the root of already cloned `dss_capi repo`
```
/dss_capi$ git clone git@github.com:dss-extensions/dss_capi.git
``` 

4. Merged contents of `klusolve/lib` into `dss_capi/lib`
```
cp -r klusolve/lib/linux_x64 lib/linux_x64
```

5. Result of running the build script
```
dss_capi$ ./build/build_linux_x64.sh
+ python3 src/classic_to_ctx.py
+ rm -rf build/units_x64
+ mkdir build/units_x64
+ fpc -Px86_64 @src/linux-x64.cfg -B src/dss_capi.lpr
Error: Illegal parameter: -vm6058
Error: /usr/bin/ppcx64 returned an error exitcode
```

I am on an intel x86 machine running Ubuntu 20.04 (if that matters).

PMeira, 2022-06-23:
Use Free Pascal 3.2.2 instead (3.0.4 is from 2017, we need 3.2 for some language features and proper ARM support). 

The original FPC installer from https://sourceforge.net/projects/freepascal/files/Linux/3.2.2/fpc-3.2.2.x86_64-linux.tar/download should work fine if the new version is not in the Ubuntu repository. Just remember to add it to your path, etc.

keegit, 2022-06-23:
Managed to figure that out - but now I'm getting this
```
/usr/bin/ld: cannot find -lklusolvex
```

I noticed #110 - ironically enough I'm trying to create Rust bindings to the C API.

PMeira, 2022-06-23:
Ah, didn't notice you used the static option for klusolvex. The build scripts/options still need some tweaks as noted in #110.

keegit, 2022-06-23:
So I followed the instructions from the SourceForge `klusolve` repository, and have the following `lib` folder structure in my local `klusolve`:
```
lib
└── linux_x64
    ├── libklusolvex.a
    └── libklusolvex.so
```

The SourceForge repository mentions linking these files, but I'm not sure how to do that properly because `libklusolvex.so` doesn't actually point to anything.

PMeira, 2022-06-24:
> instructions from the SourceForge

Don't follow those. We use a fork, it's almost completely a rewrite of the original klusolve, hence the name klusolvex to differentiate it. 

> doesn't actually point to anything.

I'm not sure I understand what you mean by that.

For a static build, you would use the `.a`, but you'd need to update the build flags in DSS C-API etc. 

If you don't need a static build, just remove `-DKLUSOLVE_LIB_TYPE=STATIC` from the CMake configuration. 

You can see how our binaries are built from these two:
- https://github.com/dss-extensions/klusolve/blob/master/ci/build_linux_x64.sh#L12-L17
- https://github.com/dss-extensions/dss_capi/blob/master/.github/workflows/builds.yml#L9

keegit, 2022-06-24:
I've got it building - but I'm having another issue. Going to open a separate one if that's alright.