dss-extensions/dss_capi #114: `Circuit_Get_Losses` segfaults

keegit, 2022-07-12:
I have the following Rust binding to the `Circuit_Get_Losses` function. Calling it produces a segmentation fault and core dump.
```rust
dss_c::Circuit_Get_Losses(result_ptr, result_cnt);
```
output:
```
Segmentation fault (core dumped)
```
I'm assuming there's some state initialization that I haven't done, but I'm unsure what needs to be initialized.

PMeira, 2022-07-12:
> I'm assuming there's some state initialization that I haven't done, but I'm unsure what needs to be initialized.

Yep. See also https://github.com/dss-extensions/dss_capi/issues/56

PMeira, 2022-07-12:
If you're using the master branch and 0.12 beta releases, please provide the sequence of calls so I can try to reproduce. 

If it's 0.10.x, avoid opening new issues for that. Although there are a few more places that need state checks (per #56), half of the code has been rewritten since 0.10.x.

Worst case scenario, link to the debug version of DSS C-API and you should get detailed info -- either started through a debugger like GDB or load the core dump afterwards. Some general info at https://github.com/dss-extensions/dss_capi/blob/master/docs/debug.md

keegit, 2022-07-13:
I'm using the master branch, here is the sequence of calls I make:
```
DSS_Start(0);
Text_Set_Command("Redirect <path to dss file>");
Solution_Set_Mode(0); // 0 for snapshot mode
Circuit_Set_ActiveElement();
Loads_Set_kW();
Loads_Set_kvar();
// conditionals
if device == reg {
  RegControls_Set_Name();
  RegControls_Set_TapNumber();
  RegControls_Set_Max_TapChange();
}
if device == cap {
  Capacitors_Set_Name();
  Capacitors_Set_States([0]);
}
Solution_Solve();
Circuit_Get_Losses(); <-- Segfault
```

PMeira, 2022-07-13:
hm... seems fine. Are you initializing the two pointers for that function?

> `Solution_Set_Mode(0); // 0 for snapshot mode`
There is an enum for that: https://github.com/dss-extensions/dss_capi/blob/cd3cf22df69e244f3ca987bdee4358597d88ac8c/include/dss_capi.h#L47-L64

I noticed you're not checking for DSS errors. Check `Error_Get_Number()` -- if that is non-zero, `Error_Get_Description()` should list the error message.

keegit, 2022-07-13:
Figured it out - the issue was that I was not properly initializing the pointers passed to the function.

closed 2022-07-13 by keegit