dss-extensions/dss_capi #121: Support for mac m1, darwin_arm64

christopinka, 2023-05-16:
Please consider adding support for mac m1. Would need a darwin_arm64 directory for starters. Thanks

PMeira, 2023-05-17:
It's already supported and tested for a few versions. See the releases page for binaries: https://github.com/dss-extensions/dss_capi/releases

PMeira, 2023-05-17:
The docs state that in a few places, e.g. from README.md:

> Through the other projects under DSS Extensions, DSS C-API enables projects to use the OpenDSS in multiple platforms (Windows, Linux, macOS) across multiple architectures (Intel x86, x86-64, ARM32, and ARM64, including Apple's M1 and M2).

closed 2023-05-17 by PMeira

christopinka, 2023-05-17:
> *** CAUTION! EXTERNAL SENDER *** STOP. ASSESS. VERIFY!! Were you expecting this email? Are grammar and spelling correct? Does the content make sense? Can you verify the sender? If suspicious report it, then do not click links, open attachments or enter your ID or password.

I'm on a project that is cloning with submodules and building every time. Do you suggest using the release asset binaries instead? Is there a recommended way to reference them in a build? In this case rust and cargo.

PMeira, 2023-05-17:
> I'm on a project that is cloning with submodules and building every time. Do you suggest using the release asset binaries instead? Is there a recommended way to reference them in a build? In this case rust and cargo.

The Pascal code doesn't benefit from custom builds, so you could use the pre-built binaries. For [klusolvex](https://github.com/dss-extensions/klusolve), it's usually worth rebuilding with custom flags on x64.

Note that, per https://github.com/dss-extensions/dss_capi/issues/110, building static libs is not an option yet. That will be an option when the C++ port is finished.

PMeira, 2023-05-25:
> Is there an example of how to link in another project?

Really depends on the language and platform, but there's nothing special since it's a shared library. 

For our bindings:

- OpenDSSDirect.jl: Julia can load the binary directly. The build step of the package downloads a tagged version.
- DSS MATLAB: MATLAB can load directly too. Specific packages are build on CI based on tags.
- DSS Sharp: same... A single package (nupkg) contains the binaries for all supported platforms there.
- dss_python_backend (for DSS-Python and ODD.py): a CFFI wrapping library is built and linked for performance and safety. On Windows, .lib is used/required for MSVC, on Linux and macOS there's no need.

For GCC Linux, there's something in the comments in the example: https://github.com/dss-extensions/dss_capi/blob/master/examples/ctx_openmp.c
The build arguments on dss_python_backend also give an idea, but it's abstracted away by the tooling there.

Using the C API directly is obviously possible (we use ourselves for some custom code), but it's cumbersome and unless the users are very advanced OpenDSS users, there's little benefit. We have an on-going effort to finish a C++ port of the Pascal code, and when that's finished I believe using C++ would be a more comfortable alternative (and maybe integrated with Conan and/or vcpkg).

> Still not example of rebuild with custom flags.

For public releases, we don't build with custom flags since the binaries wouldn't be redistributable.

> Also, not understanding comment about "For [klusolvex](https://github.com/dss-extensions/klusolve), it's usually worth rebuilding with custom flags on x64." Isn't klusolve a dependency of dss_capi? Is there somewhere that explains this? Wouldn't that mean you'd have to rebuild dss_capi as well or does dss_capi just need klusolve lib available to it.

Basically the pre-built binary is build without some specific architecture features, so they're compatible with a wider range of processors. E.g. we don't use AVX512 since the majority of desktop processors still don't support it, but most Xeon processors do. Even if the processor supports AVX512, depending on the generation of the processor it might not be worth to use it.

> Isn't klusolve a dependency of dss_capi?

It's a dynamic library, shared object, etc. It's a dependency but can be replaced on its own, without rebuilding DSS C-API, etc.  
IIRC, an optimized/custom klusolvex binary could get around 5% runtime saving. So, nothing to worry too much about, unless you're running many machines/cluster 24/7. You could always replace the shared library for klusolvex (.so/.dll/.dylib) later.

> Is there a better place for these questions? On the github 'discussions' general?

I think using the Discussions feature would be better, yes. I saw your other post there first, for example.