dss-extensions/dss_capi #122: Memory acess violation/Segfault when acessing the API

giovanniPepi, 2023-05-17:
In C, I'm getting memory corruption in both win_64 and linux_64, using version 0.13.0-dev and libs from the release pages using this example code and getting undefined behavior as it can be seem after some runs: 

[image]

[image]

Visual Studio shows this error as soon as DSS_Start(0) line is reached: 

[image]

```
#include <stdio.h>
#include <stdlib.h>
#include "include/dss_capi.h"

void RunDSS() {

	const char* version = DSS_CAPI_VERSION;
	printf("Library version: %s\n", version);

	Text_Set_Command("ClearAll");
	DSS_Start(0);

	Text_Set_Command("Clear");
	Text_Set_Command("New Circuit.REDE basekv=13.8 Bus1=REDE pu= 1  phases=3 R1=0.78 X1=0.88 R0=0.2 X0=0.3");
	Text_Set_Command("New linecode.2E70 nphases=3 R1=0.56 X1=0.3 R0=0.97 X0=1.41 C1=0 C0=0 normamps=310 unit=km");
	Text_Set_Command("New Line.trecho1 Bus1=barra2.1.2.4 Bus2=barra1.1.2.4  phases=3 linecode=2E70 length= 66.6 units=m");
	Text_Set_Command("New Load.carga3_1 bus1=barra1.1.4 phases=1 kV = 0.12 kVA=-8 pf=1 model=1  conn=wye  Vminpu = 0.1 Vmaxpu = 1.25 NumCust=1");
	Text_Set_Command("solve");
	Error_Get_Description();

	char* returnName = Bus_Get_Name();
	printf("\nName: %s\n", returnName);

	int32_t ResultCounter = 0;
	double* ResultPtrVolt = NULL;

	Bus_Get_Voltages(&ResultPtrVolt, &ResultCounter);
	printf("\nResult count: %d", ResultCounter);
	for (int i = 0; i <= ResultCounter; i += 2) {
		printf("\nBus Voltage[%d, %d]: %f, %f", i, i + 1, ResultPtrVolt[i], ResultPtrVolt[i + 1]);
	}

	int32_t* NodeResultPtr = NULL;
	int32_t NodeResultCount = 0;

	Bus_Get_Nodes(&NodeResultPtr, &NodeResultCount);
	printf("Bus Nodes: %d\n", NodeResultCount);
	for (int j = 0; j < NodeResultCount; j++) {
		printf("Node[%d]: %d\n", j, NodeResultPtr[j]);
	}
	int32_t numBus = Circuit_Get_NumBuses();
	printf("\bNum Bus: %d", numBus);

	Text_Set_Command("ClearAll");
	DSS_Get_COMErrorResults();
	Error_Get_ExtendedErrors();
	Error_Get_Description();
}
```

It seems that it's something related to how the API is handling the pointers, but I'm not quite sure yet

PMeira, 2023-05-17:
> version 0.13.0-dev 

Hm, it seems the release script is not updating the version in the header. You can use `DSS_Get_Version()` instead for the complete info.

```
	int32_t ResultCounter = 0;
	double* ResultPtrVolt = NULL;

	Bus_Get_Voltages(&ResultPtrVolt, &ResultCounter);
```

This is wrong. What is this based on?

PMeira, 2023-05-17:
```
	Text_Set_Command("ClearAll");
	DSS_Start(0);
```

This is also wrong. `DSS_Start` should always be the first call, especially if you're using threads.

By the way, we don't recommend using the C API directly. You also need to check for errors and manage the pointers with care. It's very easy to pass wrong pointers.

DSS C-API is tested extensively through the downstream projects, it's unlikely that basic point access would crash. Trying to access uninitialized results may results in some issues, but even those we've been slowly handling better in case of user error.

giovanniPepi, 2023-05-17:
> Hm, it seems the release script is not updating the version in the header. You can use `DSS_Get_Version()` instead for the complete info.
>
> ```
> 	int32_t ResultCounter = 0;
> 	double* ResultPtrVolt = [...]

I was using the parameters defined in the header dss_capi.h: 
[image]

It seems that the error was caused by using direct text commands. I've changed the previous snippet to use

```
	Text_Set_Command("compile master.dss");
	Solution_Solve();
	Error_Get_Description();

```

and the memory errors are gone

PMeira, 2023-05-17:
The errors may seem to be gone, but I assure you the root cause is not gone and it will explode down the line. You can try running through valgrind to confirm. I recommend checking the few C examples and docs we have instead of assuming a pointer is just a single value. `ResultCount` can be two or four values depending on the settings at runtime.

I don't want to be too harsh but most of your code doesn't even make any sense. What do you expect this will do?

```
	DSS_Get_COMErrorResults();
	Error_Get_ExtendedErrors();
	Error_Get_Description();
``` 

-  `DSS_Get_COMErrorResults` is used to get the current value of the `COMErrorResults` flag.
-  `Error_Get_ExtendedErrors` also returns the current value of a flag, `ExtendedErrors`.
-  `Error_Get_Description` returns the current error string.

And you're ignoring all returned values from the functions anyway.

If you haven't used OpenDSS before, I'd also recommend playing around with the official COM implementation to understand the concepts (some of them are unconventional) before dropping to a low-level API. The mapping from the official COM API to this API is very direct as mentioned in our docs.

closed 2023-05-17 by PMeira