dss-extensions/dss_capi #125: Error: signal SIGSEGV: address not mapped to object (fault address: 0x4)

keegit, 2023-11-27:
I am trying to set a storage's kW/kVAr/PF settings, but am currently getting a SIGSEGV error when trying to get all the property names of a given element. My project is in Rust but uses the bindgen crate to generate Rust bindings to the C API (all `dss_c::*` functions are the 'unsafe' C bindings). Below is my code:

```rust
pub fn get_all_property_names() -> Vec<String> {
    unsafe {
        let ctx = dss_c::ctx_Get_Prime();
        let mut result_ptr: *mut *mut *mut c_char = ptr::null_mut();
        let result_cnt: *mut i32 = ptr::null_mut();

        // SIGSEGV exception occurs immediately after this line when stepping through code
        dss_c::ctx_CktElement_Get_AllPropertyNames(ctx, result_ptr, result_cnt);
    }
}

fn test() {
    // start DSS
    assert!(dss::start(0) != 0);
    
    // load model with storage device 's680'
    assert!(dss::text_set_command("redirect tests/data/IEEE13Nodeckt.dss").is_ok());
    
    // set active element
    assert!(circuit::set_active_element("Storage.s680").is_ok());

    // get property names
    let prop = get_all_property_names();
}
```

When stepping through code - I see that the exception occurs [here](https://github.com/dss-extensions/dss_capi/blob/162eee8cacc06563577d10775d6b87a5b1f8a3f9/src/CAPI/CAPI_Utils.pas#L445)

Am I setting an element wrong? should I be using the API differently? not quite sure what the issue is.

PMeira, 2023-11-27:
I think the issue is here, it needs to be an array of two elements instead of a plain pointer:

```rust
        let result_cnt: *mut i32 = ptr::null_mut();
```

Besides `ctx_DSS_Set_EnableArrayDimensions` I mention below, there's also some notes on https://github.com/dss-extensions/dss_capi/blob/0.13.4/docs/usage.md#memory-management-functions

In an upcoming branch, I renamed `ResultCount` in the header to `ResultDims` to make things clearer:

```c
DSS_CAPI_DLL void ctx_CktElement_Get_AllPropertyNames(void* ctx, char*** ResultPtr, int32_t* ResultDims);
```

(It might be better to change this to an opaque struct someday.)

`ResultDims` since v0.10 is at least a two-element array. Element 0 is the current count, element 1 is the capacity. This allows reusing the buffer, reducing memory allocations. That is, you need to remember to call `DSS_Dispose_PPAnsiChar` after processing the buffer on `result_ptr`, since it's memory managed by the engine, but you can reuse the same buffer and call `DSS_Dispose_PPAnsiChar` once later. The `*_GR` functions basically use a buffer managed in the engine to do this. 

Recently, you can use `ctx_DSS_Set_EnableArrayDimensions` to interpret `ResultDims` as a 4-element array, where the 3rd and 4th elements represent a matrix shape:

https://github.com/dss-extensions/dss_capi/blob/0.13.4/include/dss_capi_ctx.h#L1535-L1552

Since the size is currently dynamic (2 or 4 elements), using a fixed-size array in the headers wouldn't work, hence the pointer.

Most of this buffer reusing and resizing stuff doesn't matter much for strings, since each string is a new pointer anyway (the `_GR` functions for arrays of strings will be removed by v1.0). It's more useful for ints and floats.

keegit, 2023-11-27:
[Linking this issue for relevancy](https://github.com/dss-extensions/dss_capi/issues/11l)

PMeira, 2023-11-27:
> [Linking this issue for relevancy](https://github.com/dss-extensions/dss_capi/issues/11)

@keegit Got it (extra char in the URL, fixed above), I guess using only Python for the example was a bad idea since a lot already handled behind the curtains.  
Let me give it a try, I'll update you in in the next hour.

By the way, I'm investigating generating the bindings for Go (initially only the "classic" API) per https://github.com/dss-extensions/dss-extensions/issues/33

Do you think it's worth doing the same for Rust? Certainly could use some help and feedback about it (please feel free to email me if you prefer), especially since you already have experience around it.

I was planning to investigate creating a Rust API only for the new (work-in-progress) API and after the C++ port is released due to the dynamic vs static lib preferences, but it shouldn't take too much time to start it earlier.

PMeira, 2023-11-27:
Give this a try. Created a new project from scratch using bindgen, seems to work as expected:

`main.rs`:
```rust
use std::{ffi::{c_void, c_char, CStr}, ptr::null, slice::from_raw_parts_mut};

include!(concat!(env!("OUT_DIR"), "/bindings.rs"));

fn main() {
    unsafe {
        DSS_Start(0);
        let ctx = ctx_Get_Prime();
        ctx_Text_Set_Command(ctx, "new circuit.test".as_ptr() as *const i8);
        ctx_Text_Set_Command(ctx, "new storage.we_need_this_for_the_properties".as_ptr() as *const i8);
        ctx_Circuit_SetActiveElement(ctx, "storage.we_need_this_for_the_properties".as_ptr() as *const i8);
        let mut result_dims: [i32; 4] = [0, 0, 0, 0];
        let mut result_ptr: *mut *mut c_char = 0 as *mut *mut c_char;
        ctx_CktElement_Get_AllPropertyNames(ctx, &mut result_ptr, result_dims.as_ptr() as *mut i32);
        let vec_strs = from_raw_parts_mut(result_ptr, result_dims[0] as usize).to_vec();
        for n in 0..result_dims[0] {
            println!("PropIdx:{}, PropName:{}", n + 1, CStr::from_ptr(vec_strs[n as usize]).to_str().unwrap());
        }
        println!("{} properties", result_dims[0]);
        DSS_Dispose_PPAnsiChar(&mut result_ptr, result_dims[0]);
    }
}
```

<details>
<summary>build.rs and header</summary>

`build.rs`:

```rust
use std::env;
use std::path::{Path};
use std::path::PathBuf;

fn main() {
    let pwd_dir = env::var("CARGO_MANIFEST_DIR").unwrap();
    let inc_path = Path::new(&*pwd_dir).join("../dss_capi/include");
    let lib_path = Path::new(&*pwd_dir).join("../dss_capi/lib/linux_x64");
    println!("cargo:rustc-link-search=native={}", lib_path.to_str().unwrap());
    // println!("cargo:rustc-link-lib=dylib=dss_capid");
    println!("cargo:rustc-link-lib=dylib=dss_capi");
    println!("cargo:rerun-if-changed=src/dss_capi_wrapper.h");    

    // https://rust-lang.github.io/rust-bindgen/tutorial-3.html
    let bindings = bindgen::Builder::default()
        .clang_arg(format!("-I{}", inc_path.to_str().unwrap()))
        .header("src/dss_capi_wrapper.h")
        // .parse_callbacks(Box::new(bindgen::CargoCallbacks))
        .prepend_enum_name(false)
        .generate()
        .expect("Unable to generate bindings");

    // Write the bindings to the $OUT_DIR/bindings.rs file.
    let out_path = PathBuf::from(env::var("OUT_DIR").unwrap());
    bindings
        .write_to_file(out_path.join("bindings.rs"))
        .expect("Couldn't write bindings!");    
}

```

`dss_capi_wrapper.h`:

```
#include <stddef.h> //TODO: add to DSS C-API headers for size_t
#include "dss_capi.h"
#include "dss_capi_ctx.h"
```

...plus added to `Cargo.toml`:

```
[build-dependencies]
bindgen = "0.69.1"`
```

</details>

keegit, 2023-11-27:
> Do you think it's worth doing the same for Rust? Certainly could use some help and feedback about it (please feel free to email me if you prefer), especially since you already have experience around it.

I'm not a power engineer - so definitely take my opinion with a grain of salt. I _definitely_ think it's worth it. A lot of utilities seem to value it's ability to eliminate common memory bugs.

I will also say that builds are a bit tricky due to a couple factors:

- in our project, we build the `klusolve` and `capi` and link them as shared libs. We use the `cmake` crate to build these projects, but one of the dependencies of klusolve causes the builds to fail unless it's downloaded every time. I think ideally you would assume your system has _all_ shared libs pre-installed, and then you can just link them.

- The linking of the shared libs is required by the project that generates the bindings, so any project that uses the crate will require a build script that links them.

If you need help doing so feel free to message me, would be happy to help if possible.

keegit, 2023-11-28:
@PMeira So I've got things working, except when I try to set kW, it doesn't get set to that value? not sure if that could be a consequence of the model we're running. (I'm not a power engineer)

```rust
pub fn set_kw(value: f64) -> Result<()> {
    unsafe {
        let ctx = dss_c::ctx_Get_Prime();
        if ctx == ptr::null_mut() {
            return Err(DssError::NullCPtr);
        }
    
        let element_idx = dss_c::Storages_Get_idx();
        let class_idx = dss::set_active_class("Storage")?;

        let element_ptr = dss_c::Obj_GetHandleByIdx(ctx, class_idx, element_idx);

        if element_ptr == ptr::null_mut() {
            return Err(DssError::NullCPtr);
        }

        let property_idx = (ckt_element::get_all_property_names().iter().position(|p| p == "kW").unwrap() + 1) as i32;
        dss_c::Obj_SetFloat64(element_ptr, property_idx, value);
    }
    Ok(())
}

pub fn get_kw() -> Result<f64> {
    unsafe {
        let ctx = dss_c::ctx_Get_Prime();
        if ctx == ptr::null_mut() {
            return Err(DssError::NullCPtr);
        }

        let element_idx = dss_c::Storages_Get_idx();
        let class_idx = dss::set_active_class("Storage")?;

        let element_ptr = dss_c::Obj_GetHandleByIdx(ctx, class_idx, element_idx);
        if element_ptr == ptr::null_mut() {
            return Err(DssError::NullCPtr);
        }

        let property_idx: i32 = ckt_element::get_all_property_names().iter().position(|p| p == "kW").unwrap().try_into().unwrap();
        let kw = dss_c::Obj_GetFloat64(element_ptr, property_idx.try_into().unwrap());
        Ok(kw)
    }
}

#[test]
fn set_storage() {
    assert!(dss::start(0) != 0);
    assert!(dss::text_set_command("redirect tests/data/IEEE13Nodeckt.dss").is_ok());
    assert!(circuit::set_active_element("Storage.s680").is_ok());
  
    // set kw to 6.0
    dss_rs::storages::set_kw(6.0).expect("Failed to set kW for Storage.s680");
    dss::solution_solve();
    
    // kw ends up being -0.25 ?
    println!("storage kW: {:?}", dss_rs::storages::get_kw());
}
```

```bash
running 1 test
storage kW: Ok(-0.25)
test set_storage ... ok

test result: ok. 1 passed; 0 failed; 0 ignored; 0 measured; 8 filtered out; finished in 2.67s
```

closed 2023-11-28 by keegit

PMeira, 2023-11-28:
> Edit: Do I need to set kWrated instead?

I guess you managed to make it work, right? The other properties indeed need to allow for the kW value, plus the current charge level.

Regarding the API usage, if you need some tests, editing through the text interface (`Text_Set_Command("Storage.s680.kW=6.0")`) should result in the same thing but slower, especially if the class and property indexes are cached.

> I definitely think it's worth it. A lot of utilities seem to value it's ability to eliminate common memory bugs.

Yeah, regardless of the language, a higher level interface that hides most of the pointer madness is great, and Rust itself would certainly add to it.  
Per the golang ticket, I'll try to implement the basics and generate the rest for the classic API by Dec 11 and will take opportunity to the same for Rust.

> If you need help doing so feel free to message me, would be happy to help if possible.

I'll be sure to mention you @keegit so you can have a look later!

After that, if the new API works fine in the initial Python release, the plan is to add it to most other language bindings too. It's **much** simpler to use and will cover all DSS object classes as I mentioned before.

Sooner or later EPRI will have an official Linux version too, so the implementation for the classic API will still be useful in some scenarios. I'm also working on a wrapper that exposes the official binary as DSS C-API (only Windows for now).

PMeira, 2023-12-18:
Just created an issue to track the initial Rust bindings at https://github.com/dss-extensions/dss-extensions/issues/34 and the empty repo at https://github.com/dss-extensions/AltDSS-Rust (@keegit)

Besides CI and building scripts, the initial [Golang impl.](https://github.com/dss-extensions/AltDSS-Go/) is done and works fine. I imagine we will reach the same or better status for Rust, potentially during this week, or right after the holidays. Some of the work done to achieve the Go bindings will make the Rust bindings better/easier.