dss-extensions/dss_capi #130: Memory leak

keegit, 2023-12-21:
I actually think this is because I am not handling memory propery in my use of the API versus there actually being a bug in it, but either way I could use some help.

We're currently using the C API within a genetic algorithm - so there are thousands of API calls happening over the course of the algorithm. Using valgrind, we've identified the C API to be the source of the memory leak. I've included the valgrind output as an attached file.

My question is - do I need to be releasing memory for all API calls? or just a few. I notice in the output that leaks occur in functions I don't necessarily call.

Any help is greatly appreciated.

PMeira, 2023-12-22:
There were some tiny leak when the state was invalid and some for empty data. These were fixed in https://github.com/dss-extensions/dss_capi/commit/41295d04670f8b342e85f4e8689252acd76d328a, which is included in v0.14.0b1 (check the pre-releases). 

I'd need to investigate something more specific another time (not for a few days at least). I also can't make any sense of those traces, maybe try linking with the debug version (`libdss_capid.so`) for better traces.

Valgrind has some false-positives with Free Pascal, but that's a fixed set of potential leaks that doesn't grow, related to the FPC runtime initialization and things like that, AFAICT.

If you're not using the `*_GR` functions, the array buffers can be reused and freed once (with the appropriate API functions, not directly). Functions from the classic API that return single strings keep a copy of the last string in the engine too, so you don't need to free those -- just copy them before another API call that returns a string since it could overwrite it (Free Pascal strings are refcounted, so it depends).

For the `*_GR` alternatives, there are internal buffers (that grow as required) for each basic type (notably arrays of float64, int32 and int8; deprecated for strings).

All buffers are specific to each DSS context, there are no global buffers at process level anymore. You could use `(ctx_)DSS_ResetStringBuffer` and `(ctx_)DSS_DisposeGRData` to dispose the internal buffers, but those don't leak and should be disposed automatically when the context itself is disposed (in case of the prime/default instance, that happens when the library is finalized.

For the new APIs, some functions return strings that need to be manually disposed. Most try to expose internal state directly and don't require disposal.

For nearly all functions, any pointer provided by the user must be handled by the user, e.g. if you provide a string to `ctx_Circuit_SetActiveElement`, it doesn't take ownership of the string (same is valid for almost all arrays).

keegit, 2023-12-22:
So linking to the debug version shows that there are only two places where the memory leaks are thought to occur:
- `text_set_command()`
  - this is _likely_ due to my mishandling of the string passed into the function, but I'm still investigating.
- `solution_solve()`
  - I'm going to update the version I'm using to the one you linked. Hopefully that eliminates `solve()` as the source.

PMeira, 2024-01-17:
Per the full valgrind log and the snippets here, I think there was probably a memory corruption somewhere.  

I modified the current [ieee13.rs example](https://github.com/dss-extensions/AltDSS-Rust/blob/8e688811fb73af758ff80f3bf8d99ccb2e32757c/examples/ieee13.rs) to include and run this function instead:

```rust
fn test_regcontrols(dss: &IDSS) -> Result<(), DSSError> {
    let solution = &dss.ActiveCircuit.Solution;
    let reg_controls = &dss.ActiveCircuit.RegControls;
    use altdss::classic::{SolveModes, ControlModes};

    run_ieee13(&dss)?;

    // Adjust the loads and solution mode so we get tap changes
    solution.Set_Mode(SolveModes::Daily)?;
    solution.Set_ControlMode(ControlModes::Time)?;
    solution.Set_Number(1)?;
    solution.Set_LoadMult(1.3)?;
    dss.Command("BatchEdit Load..* Daily=Default".to_string())?;

    const NUM: i32 = 100000;
    let mut total_changes = 0;
    for i in 0..NUM {
        let mut idx = reg_controls.First()?;
        let val = 1 + (i % 2);
        loop {
            if idx == 0 { 
                break; 
            }
            // set it to 1 or 2 depending on the current index
            reg_controls.Set_MaxTapChange(val)?;
            total_changes += 1;
            idx = reg_controls.Next()?;
        }
        solution.Solve()?;
    }
    println!("Total number of Set_MaxTapChange calls: {total_changes}");
    Ok(())
}
```

Since there is a `Solve()` call, it takes a while to run through valgrind, but seems clean:

```
Total number of Set_MaxTapChange calls: 300000
<snip>
==29056== LEAK SUMMARY:
==29056==    definitely lost: 0 bytes in 0 blocks
==29056==    indirectly lost: 0 bytes in 0 blocks
==29056==      possibly lost: 0 bytes in 0 blocks
==29056==    still reachable: 74,366 bytes in 5 blocks
==29056==         suppressed: 0 bytes in 0 blocks
==29056== 
==29056== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

Let's close this for now. It anything remains we can reopen later.

closed 2024-01-17 by PMeira