dss-extensions/dss_capi #132: dss_capi can not be used on ubuntu20.4

ericwoo-github, 2024-06-21:
when built on ubuntu20.4, dss_capi can not be used with the error

PMeira, 2024-06-21:
About the errors you listed, either you didn't download the binaries or the build scripts failed. If you are indeed building, you need Free Pascal 3.2.2, currently. The system packages could be outdated etc., but Free Pascal can be manually installed. As a reminder, if you are building from scratch, you also need to build KLUSolveX (tag 1.0.0a1 is used for the current binaries), and that needs C++ and C compilers, and CMake.

Have you tried the pre-built binaries? All downstream projects here on DSS-Extensions that run on GitHub Actions use Ubuntu in one way or another.

We don't explicitly support WSL, but I don't see any specific reason for it to fail there.

As a reminder, we don't recommend new users to call DSS C-API directly, since we already do the work of integrating it to various programming languages. More at https://dss-extensions.org/

ericwoo-github, 2024-06-22:
1. fpc is already “Free Pascal Compiler version 3.2.2”
2. KLUSolveX also built with libklusolvex.so installed to /usr/local/lib

ericwoo-github, 2024-06-22:
1.fpc is already “Free Pascal Compiler version 3.2.2”
2.KLUSolveX built successfully with libklusolvex.so generated
3.dss_capi also built successfully with libdss_capi.so and libdss_capid.so generated
it just does not work, i am so much confused

PMeira, 2024-06-22:
Probably nothing to do we DSS C-API, so I'm closing this. 

Just tested the following script with a fresh Ubuntu 20.04 install:

```
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

$ uname -a
Linux altdss 5.4.0-186-generic #206-Ubuntu SMP Fri Apr 26 12:31:10 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

$ gcc --version
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```bash
# git is preinstalled in the default server installation, included here just to be sure
sudo apt install g++ cmake git 

# ---Alternative 1: Using our pre-built binaries

cd
mkdir using_prebuilt
cd using_prebuilt
git clone --branch 0.14.5 https://github.com/dss-extensions/dss_capi
# Clone the example circuits
git clone https://github.com/dss-extensions/electricdss-tst

wget https://github.com/dss-extensions/dss_capi/releases/download/0.14.5/dss_capi_0.14.5_linux_x64.tar.gz
tar zxfv dss_capi_0.14.5_linux_x64.tar.gz
cd dss_capi/examples
gcc -o minimal_gr minimal_gr.c -I../include -L../lib/linux_x64 -ldss_capi -lklusolvex

# Copy a circuit here
cp ../../electricdss-tst/Version8/Distrib/IEEETestCases/13Bus/* .
ln -s ./IEEE13Nodeckt.dss master.dss

LD_LIBRARY_PATH=`pwd`/../lib/linux_x64 ./minimal_gr

# --- Alternative 2: Rebuilding
cd

mkdir custom_build
cd custom_build

wget -O fpc.tar https://sourceforge.net/projects/freepascal/files/Linux/3.2.2/fpc-3.2.2.x86_64-linux.tar/download
tar xf fpc.tar
cd fpc-3.2.2.x86_64-linux
./install.sh < /dev/null # accept all defaults
cd ..

git clone --branch 0.14.5 https://github.com/dss-extensions/dss_capi

# Build KLUSolveX
git clone --branch 1.0.0a1 https://github.com/dss-extensions/klusolve
cd klusolve
# Since the old CMakeLists uses the now dead BitBucket link, let's manually download
# (the master branch already has updated links, but we're reproducing the same as the pre-built binaries here)
wget https://gitlab.com/libeigen/eigen/-/archive/3.3.7/eigen-3.3.7.tar.gz
tar zxf eigen-3.3.7.tar.gz
mv eigen-3.3.7 eigen-eigen-323c052e1731
cmake . -DUSE_SYSTEM_SUITESPARSE=OFF
cmake --build . --config Release
mv lib/linux_x64/* ../dss_capi/lib/linux_x64/

# Build DSS C-API
cd ../dss_capi
export PATH=$PATH:~/fpc-3.2.2/bin

# Now finally build one example
cd examples
gcc -o minimal_gr minimal_gr.c -I../include -L../lib/linux_x64 -ldss_capi -lklusolvex

# Copy a circuit here
cp ../../electricdss-tst/Version8/Distrib/IEEETestCases/13Bus/* .
ln -s ./IEEE13Nodeckt.dss master.dss

LD_LIBRARY_PATH=`pwd`/../lib/linux_x64 ./minimal_gr
```

closed 2024-06-22 by PMeira

PMeira, 2024-06-22:
Updated my previous message with some missing commands to ensure it works as-is in a fresh environment. Copying and pasting the work script should work fine now.

ericwoo-github, 2024-06-23:
it works with "Alternative 2: Rebuilding"
thanks so much