dss-extensions/dss_capi #135: Rename repository

PMeira, 2024-10-22:
Most likely to "AltDSS_Engine" or something like that. Update references in the docs accordingly.

HHuuuhh666, 2024-12-30:
Hi,PMeira
Why is there such a big difference in the execution results between x64 and x86 for OpenDSS. exe?
Here is my DSS.

Clear
New Circuit.Circuit1 Phases=3 bus1=Bus1.1.2.3 basekv=220 pu=1.0 angle=0 MVAsc3=11297 MVAsc1=11297 x1r1=10 x0r0=10
New Capacitor.Filter1 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=93.37 XL=3.94 R=0
New Capacitor.Filter2 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=119.76 XL=1.72756 R=0
New Capacitor.Filter3 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=119.76 XL=1.10552 R=0
New Line.Line1 bus1=Bus1.1.2.3 bus2=Bus2.1.2.3 Length=4.26 units=km R1=0.054 X1=0.308 C1=0 R0=0.054 X0=0.308 C0=0
New LineCode.Line1 Nphases=3 Rmatrix=[0|0 0|0 0 0] Xmatrix=[0|0 0|0 0 0]
New Load.Load1 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 kVA=155000 Pf=0.78 rneut=0 xneut=0 model=1
New Load.Load2 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 kVA=26000 Pf=0.85 rneut=0 xneut=0 model=1
New Load.Load3 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 Kw=0 kvar=180000 rneut=0 xneut=0 model=1
New Capacitor.PCap1 Phases=1 bus1=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68]
New Capacitor.PCap2 Phases=1 bus1=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68]
New Capacitor.PCap3 Phases=1 bus1=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68]
New Reactor.PRea1 Phases=1 bus1=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0]
New Reactor.PRea2 Phases=1 bus1=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0]
New Reactor.PRea3 Phases=1 bus1=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0]
New Capacitor.SCap1 Phases=1 bus1=Bus3.1.0 bus2=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933]
New Capacitor.SCap2 Phases=1 bus1=Bus3.2.0 bus2=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933]
New Capacitor.SCap3 Phases=1 bus1=Bus3.3.0 bus2=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933]
New Transformer.T2w1 Phases=3 XHL=13 %imag=0 %Noloadloss=0 Windings=2 Wdg=1 bus=Bus2.1.2.3.0 Conn=Wye Kv=220 Kva=160000 %R=0 Tap=1 rneut=0 xneut=0 Wdg=2 bus=Bus3.1.2.3 Conn=Delta Kv=35 Kva=160000 %R=0 Tap=1
set DefaultBaseFrequency=50
set Basefrequency=50
set Frequency=50
set Algorithm=Newton
set Maxiter=50
set voltagebases=[220.0, 35.0, 20.2]
calcvoltagebases
solve
export voltages

PMeira, 2025-10-30:
Probably going with AltDSS-Core due to the various components being published here in the future:

- (Alt)DSS C-API
- AltDSS engine/library
- Oddie, which exposes EPRI's binaries as AltDSS C-API
- DSS-Extensions COM Bridge
- AltDSS.hpp
- User models and a few examples